ucb-bar/chipyard #738: Environment Scripts Missing LIBRARY_PATH

SaltedFishLZ, 2020-12-12:
**Impact**:  software

I think the env*.sh (e.g., `env. sh`, `env-riscv-tools. sh`) all have a serious bug that doesn't contain `LIBRARY_PATH`, which will make `ld` fails to some static libraries like `fesvr.a`. It will cause the building of verilator simulators to fail. Also, sometimes we need `${CPATH}` for included files.

I think if we want to use such scripts to help users set up environments, it is better to test such scripts. Otherwise, I think it is better to provide some images or provide your environment variables directly. Or we can change the Makefiles like Rocket Chip, which only requires 1 environment variable `${RISCV}`

**Tell us about your environment:**
*Chipyard Version:*  1.1.0 
*OS:*  Docker Ubuntu 18.04
*Other:* 

**Other information**

```
root@1fb33438b35a:/chipyard# cat env-riscv-tools.sh
export CHIPYARD_TOOLCHAIN_SOURCED=1
export RISCV=/chipyard/riscv-tools-install
export PATH=${RISCV}/bin:${PATH}
export LD_LIBRARY_PATH=${RISCV}/lib${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}

```

a0u, 2020-12-12:
If using the provided flow to build Verilator/VCS simulators, setting `LIBRARY_PATH` is unnecessary as libfesvr is linked by specifying the full path.

https://github.com/ucb-bar/chipyard/blob/cca3cd087c1c5a54f98867bc4a89129c1c299b94/sims/verilator/Makefile#L64
https://github.com/ucb-bar/chipyard/blob/64632c8aeef5386b4840e324be4429c0ceaac635/vcs.mk#L14

On the `dev` branch, this has been refactored into a combination of `-L` and `-lfesvr` to an equivalent effect.
https://github.com/ucb-bar/chipyard/blob/70d41996dd738cd4b32e2e845644d62c9fbe14be/sims/common-sim-flags.mk#L17-L22

Likewise, `-I$(RISCV)/include` takes care of the search path for the fesvr headers, obviating the need for `CPATH`.

https://github.com/ucb-bar/chipyard/blob/64632c8aeef5386b4840e324be4429c0ceaac635/sims/verilator/Makefile#L49
https://github.com/ucb-bar/chipyard/blob/64632c8aeef5386b4840e324be4429c0ceaac635/vcs.mk#L7

On `dev`:
https://github.com/ucb-bar/chipyard/blob/70d41996dd738cd4b32e2e845644d62c9fbe14be/sims/common-sim-flags.mk#L10

SaltedFishLZ, 2020-12-13:
Got it! Thanks for your quick response. Hope this fix can be merged into
master or any release as soon as possible.
I am trying to use the 4 releases and following the tutorials on readdocs.io
to set up the development framework. But all failed. It took me 2 days to
figure out what happened in the initialization repo. It is quite frustrating

SaltedFishLZ, 2020-12-13:
Another interesting problem is that if I choose to build the repo inside a
Dockerfile, building libgloss will fail. But it seems if I use an
interactive bash terminal even in a Docker container, it is OK.

a0u, 2020-12-13:
> Hope this fix can be merged into master or any release as soon as possible.

This is already how the current release handles things.

If you have actually encountered a compile or link error that is not the result of some errant modification, then you need to provide the full Verilator invocation and error message(s) from the build log.

a0u, 2020-12-13:
Chipyard v1.1.0 is quite old now, but linking against libfesvr shouldn't be a problem unless  `LDFLAGS` is being overridden in the make command.
https://github.com/ucb-bar/chipyard/blob/7a6124d462b211f7123b1307a27aa125cf90134b/sims/verilator/Makefile#L49

SaltedFishLZ, 2020-12-13:
Thanks for the clarification. I chose to use v1.1.0 because previously I
have an error in building the toolchain (libgloss). I am trying to debug.
It seems libgloss is incompatible with my build flags
export CC=gcc-5 CXX=g++-5

a0u, 2020-12-13:
Ah, `export CC=gcc-5 CXX=g++-5` is the crucial piece of information that explains #735.  Let's keep the discussion of libgloss-htif to that issue.

SaltedFishLZ, 2020-12-13:
Thanks. It is quite annoying to deal with some legacy codes as workloads.
If I don't use gcc-5, I have to dig deep into the Makefiles and add c
standard flags. Any suggestions?

a0u, 2020-12-13:
> If I don't use gcc-5, I have to dig deep into the Makefiles and add c standard flags.

Is there a particular reason to use the GCC 5 release series for Chipyard?  Ubuntu 18.04 LTS provides GCC 7.4.0 by default, and I can confirm that 7.5.0 works on some of our Ubuntu build hosts.

SaltedFishLZ, 2020-12-13:
Yes, definitely. I am using Ubuntu 19.04 now.
Some legacy codes like old SPEC benchmarks cannot be compiled by GCC 7.4.0 due to various Werrors.

a0u, 2020-12-13:
At least, `-Wno-error` will make all warnings non-fatal, which often helps with non-standards-conforming code.  (Functional correctness is another matter.)

SaltedFishLZ, 2020-12-13:
Thanks.
BTW, it seems on release 1.3.0, there is still no LIBRARY_PATH and CPATH in
env-riscv-tool.sh
Do I have some misunderstanding about which script to use?

Also, for 1.1.0, it seems Boom RV32 doesn't work.

a0u, 2020-12-13:
Referring back to my first comment, the `-L` and `-I` compiler flags already extend the gcc search paths to find libfesvr when building VCS/Verilator, therefore serving the same purposes as `LIBRARY_PATH` and `CPATH`.  Explicit command-line options are generally preferred over environment variables to avoid any unintended effects on whatever else the user may be doing unrelated to the simulator builds.

What is the bug?  To my knowledge, no one else has reported a linkage error with the Verilator flow, and CI is passing, so I must assume there must be something different about your setup, but you have not yet provided a build log or specific details to reproduce an error.

closed 2021-01-30 by alonamid