ucb-bar/chipyard #744: How many cores does the rocket support at most?

First-Killer, 2020-12-22:
I just read the [Technical Reports](https://www2.eecs.berkeley.edu/Pubs/TechRpts/2016/EECS-2016-17.html) but I don't know the greatest amount of cores.

lsteveol, 2020-12-22:
I don't know for sure if there even is a maximum, but you'll likely be limited by other factors such as area, power, process yield, and RAM usage to even build the RTL.

First-Killer, 2020-12-24:
Thank you very much!

And here is the answer of the author,the maximum number is 16 now:

The main scalability limitation of a tightly coupled cluster of cores is
the timing of the L1-core interconnect. Currently, we taped out and tested
PULP chips with up to 16 cores (http://asic.ethz.ch/2020/Dustin.html). Keep
in mind that the main target of PULP is low-power extreme edge computing,
so the benefit of going much beyond that in symmetric multi-processing is
probably overrun by the limitations.
That said, people in our group are working on larger scale applications
(e.g., @suehtamacv <https://github.com/suehtamacv>
https://arxiv.org/pdf/2012.02973.pdf ) so stay tuned.

First-Killer, 2020-12-24:
> 我不确定是否甚至有最大值，但是您可能会受到其他因素的限制，例如面积，功耗，工艺良率以及甚至无法构建RTL的RAM使用量。

Thank you!

closed 2021-01-25 by alonamid