ucb-bar/chipyard #746: Simulation with BOOM SoC for a long time

baichen318, 2020-12-25:
Hi all,

My simulation with the BOOM SoC seems to be stuck and I do not know why.

I have synthesized the BOOM SoC with SmallBoomConfig from ChipTop and now I compile the netlist file with other simulation backend files (e.g., TestDriver.v, SimSerial.cc, etc.) provided by Chipyard with VCS in order to simulate the dhrystone.riscv benchmark. However, it seems that it runs for a long time and does not exit. I cannot figure it out. I use ASAP7 with SRAM macros provided by Hammer (i.e., https://github.com/ucb-bar/hammer/tree/9d83bbadc0caaa7f81b4929c4e32333fc5a8d900/src/hammer-vlsi/technology/asap7) to synthesize the BOOM. My genus version is 18.13-s027_1 and my VCS version is M-2017.03. I use the command `make sim-syn MACROCOMPILER_MODE='-l /path/to/chipyard/vlsi/hammer/src/hammer-vlsi/technology/asap7/sram-cache.json --mode strict -hir chipyard.Testharness.SmallBoomConfig.top.mems.fir' CONFIG=SmallBoomConfig BINARY=/path/to/dhrystones.riscv` to synthesize and simulate.

Enclosed please find my screenshot about the simulation situation.

I also use other small benchmarks but nothing changed.

Many thanks for your help!

Best regards,
Chen

[image]

baichen318, 2020-12-26:
Hi all,

I checked that it is due to the time out issue. However, I do not know why it is timeout.

Enclosed please find the screenshot of the timeout issue.

Thank you very much.

Best regards,
Chen

[image]

jerryz123, 2020-12-26:
Thanks for reporting this, I'll take a look soon.

lsteveol, 2020-12-26:
I see in the log the CPU/DRAM freq is 1Hz, and the sim times out after 10ms. Is there anyway for you to increase the clock freq?

baichen318, 2020-12-28:
> I see in the log the CPU/DRAM freq is 1Hz, and the sim times out after 10ms. Is there anyway for you to increase the clock freq?

Hi @lsteveol ,

I have double checked my flow. **From my point of view, it is due to the SRAM Macro provided by Hammer.**

I use RTL with SRAM Macro provided by Hammer to simulate directly. However, the same circumstance goes into this case.

In conclusion, SRAM Macro causes the timeout issue.

I do not figure it out why SRAM Macro causes this issue. Could you please give some hints?

Thank you very much!

Best regards,
Chen

baichen318, 2020-12-28:
> Thanks for reporting this, I'll take a look soon.

Hi @jerryz123 

Thank you for your attention.

I think it is due to the SRAM Macro provided by Hammer. However, I do not find the root cause in SRAM Macro.

Best regards,
Chen

baichen318, 2020-12-28:
Hi all,

I have checked the waveform of SRAM Macro compared with flip-flops. I use the ISA workload rv32ui-p-addi to test. I find that the first 1803 ns works well. However, the CSB is always invalid in later tests, which cannot make SRAM Macro work well.

I was wondering what kinds of reasons to make CSB always invalid?

jerryz123, 2020-12-28:
The ASAP7 SRAM behavioral models appear to be X-initialized. SRAMs in the standard designs need to be initialized for simulation to behave correctly, else X-pessimism will break the simulation.

The behavioral models are defined here: https://github.com/ucb-bar/hammer/blob/master/src/hammer-vlsi/technology/asap7/sram_compiler/memories/behavioral/sram_behav_models.v

baichen318, 2020-12-29:
> The ASAP7 SRAM behavioral models appear to be X-initialized. SRAMs in the standard designs need to be initialized for simulation to behave correctly, else X-pessimism will break the simulation.
>
> The [...]

Hi @jerryz123 ,

Thank you for your reply.

I have checked the verilog file: sram_behav_models.v. Do you mean that I should add some initialization codes to the file in order to make SRAM Macro work well? Just like flip-flops?

Enclosed please find the verilog code snippet of flip-flops.  In the code snippet, flip-flops use $random to initialize. Should I add simliar codes to sram_behav_models.v?

Thank you very much!

[split_cc_dir_ext.v.txt](https://github.com/ucb-bar/chipyard/files/5749292/split_cc_dir_ext.v.txt)

jerryz123, 2020-12-29:
Yeah something like that should work.

jerryz123, 2020-12-29:
If you implement the memory initialization, we would appreciate a PR with that feature into hammer.

baichen318, 2020-12-29:
> If you implement the memory initialization, we would appreciate a PR with that feature into hammer.

Hi @jerryz123 ,

I have checked the inialization method. From my point of view, it would not bring any change if I just add the inialization method into sram_behav_models.v. I see that SRAM Macro cannot output a signal (i.e., auto_out_d_valid in the module `SerialAdapter` of *.top.v) to drive the serial adapter so that the serial adapter can load benchmarks into the CPU.

I'd better try to make SRAM Macro compatible with the CPU.

Best regards,
Chen

jerryz123, 2020-12-29:
The SRAMs are just memories. The bad auto_out_d signals you are witnessing are a symptom of uninitialized memories. No modifications to RTL outside of the SRAM behavioral model is needed to make GL simulation work.

Why do you think adding the initialization into sram_behav_models.v will have no effect?

baichen318, 2020-12-30:
> The SRAMs are just memories. The bad auto_out_d signals you are witnessing are a symptom of uninitialized memories. No modifications to RTL outside of the SRAM behavioral model is needed to make GL [...]

I have added the initialization stage to sram_behav_models.v and I re-simulate with the workload rv32ui-p-addi again. However, the same timout issue appears again.

Enclosed please find my modified sram_behav_models.v. From my point of view, it may not work to only add the initialization stage to those SRAM Macro.

[sram_behav_models.v.txt](https://github.com/ucb-bar/chipyard/files/5752767/sram_behav_models.v.txt)

jerryz123, 2020-12-30:
Can you send me the waveforms for that run? Its unclear to me what could be causing the issue now

baichen318, 2020-12-30:
> Can you send me the waveforms for that run? Its unclear to me what could be causing the issue now

Yeah, I have put the output vcd file in the link: https://drive.google.com/file/d/1FXoj2mUes4dKR_-GTttdWh68uTWwoU3h/view?usp=sharing

baichen318, 2020-12-30:
> > Can you send me the waveforms for that run? Its unclear to me what could be causing the issue now
> 
> Yeah, I have put the output vcd file in the link: https://drive.google.com/file/d/1FXoj2mUes4dKR_-GTttdWh68uTWwoU3h/view?usp=sharing

Hi @jerryz123 

According to the vcd file, it seems that SRAM Macro do not have valid RE and WE signals even we have valid CSB and WEB signals, which hinder SRAM Macro from working well.

baichen318, 2020-12-30:
**Update**

Hi all,

I have modified SRAM Macro provided by Hammer so that I can use those SRAM Macro to simulate. 

This time, there is no issue at all and everything goes well. Finally, I can get the entire vcd file successfully.

Enclosed please find my sceenshot which shows that simulation goes well.

Best regards,
Chen

[image]

closed 2020-12-31 by baichen318