ucb-bar/chipyard #758: `-hex` `run-binary` commands differ from normal

abejgonzalez, 2021-01-14:
**Impact**: unknown

**Tell us about your environment:**
*Chipyard Version:* https://github.com/ucb-bar/chipyard/commit/d1d7bb8f527b000e9f317f36416ca03503616d3b
*OS:* Ubuntu 18.04.5
*Other:* Using a Gemmini configuration and Rocket as follows:

Gemmini Configuration:
```
defaultConfig.copy(meshRows = 8, meshColumns = 8, dataflow = Dataflow.WS)
```

Top-Level Chipyard Configuration:
```
class GemminiEE290Lab2RocketConfig extends Config(
  new gemmini.SpecialConfig ++
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++
  new chipyard.config.AbstractConfig)
```

(Note: This was discovered in the Berkeley EE290-2 class)

**What is the current behavior?**
There is different behavior for `run-binary-fast-hex` and `run-binary-fast`. The `hex` version fails https://github.com/ucb-bar/gemmini-rocc-tests/blob/ee290-sp21-lab2/ee290/large_matmul.c while the non-`hex` version passes. It looks like some things are being cleared in memory?

**What is the expected behavior?**
Both the `-hex` and non-`hex` version of the `run-binary` commands should match.

**Other information**

Failing test:
[image]

abejgonzalez, 2021-01-14:
Adding this issue to look into at some point.

jerryz123, 2021-01-14:
Are you saying that part of the binary isn't being loaded or cleared correctly? Its possible there's still a bug in the elf-to-hex script. I ran into this issue before.
https://github.com/ucb-bar/chipyard/commit/d958b8e1aab749161f92bfeac954f9d49fd3abc1

jerryz123, 2021-01-14:
Although based on the output, is it possible that this is a gemmini bug that is exposed by the timing differences between the normal and the -hex commands?

colinschmidt, 2021-01-14:
I agree with Jerry, on the two possible causes.

The elf-to-hex script bugs are usually an entire section not getting loaded and so that seems unlikely to lead to a partially correct answer.

abejgonzalez, 2021-01-14:
Yea. I would need to look into it more but it looks like this happens with more tests.

This issue is mainly tracking the issue so that it gets looked at in the future (probably by me when I have time)

abejgonzalez, 2021-01-15:
After a further look the Mesh output looks identical from `-hex` to non `-hex` versions. So this seems like a Gemmini bug (since the same data looks to be entering/exiting the PE's).

jerryz123, 2021-01-15:
What version of gemmini are you using? Some others have been chasing down a gemmini nondeterminism bug for months

abejgonzalez, 2021-01-15:
This is using the `ee290-sp21-lab2` branch on Gemmini with a proper Verilog MeshBlackbox file (this is the ee290-2 class). I can provide a `vpd` / simulator / file that shows this.

abejgonzalez, 2021-01-15:
Nevermind this looks like an uncore issue. Gemmini is outputting the right values to memory then later down the line when Rocket requests the same memory addr, it returns the wrong values. So maybe this is an L2 issue.

abejgonzalez, 2021-01-17:
I had some SW fixes I needed to do and now this issue seems to have disappeared. So I will mark this as closed until this comes up again.

closed 2021-01-17 by abejgonzalez