ucb-bar/chipyard #760: How to link a blackbox to memory node and cbus?

flp2io, 2021-01-14:
Hi,

I want to attach a verilog module (blackbox) to the rocketchip. 
To hook up the module to the cbus I followed chipyard-docs instructions(https://chipyard.readthedocs.io/en/latest/Customization/MMIO-Peripherals.html), which works fine. However, I also want to attach the module to the dut.memAXI4Node.in(more precisely, between the memAXI4Node and mem.io_axi4.head(SimAXIMem)) at the same time. The main problem for me is that the blackbox module can not be used in object SimAXIMem to connect the wires to the node, since I have already instantiated it in the MMIO trait.
`GCD.scala`
`     val impl = Module(new GCDMMIOBlackBox(params.width))`

```
**SimAXIMem.scala**
def connectMem(dut: CanHaveMasterAXI4MemPort)(implicit p: Parameters): Seq[SimAXIMem] = {
    dut.mem_axi4.zip(dut.memAXI4Node.in).map { case (io, (_, edge)) =>
      val mem = LazyModule(new SimAXIMem(edge, size = p(ExtMem).get.master.size))
      Module(mem.module).suggestName("mem")
      //not working
      impl.io.s_axi_awid := io.aw.bits.id
      ...
```

How can my goal be achieved?

Thanks.

abejgonzalez, 2021-01-30:
Sorry, Im not sure I follow. You have a new module (call it `ModuleX`) that can connect to the CBUS. But now you want the module to also sit between the `SimAXIMem` and the AXI interface coming off the chip? So something like this:

CBUS <-> `ModuleX` 
and
AXI Interface coming off the chip <-> `ModuleX` <-> `SimAXIMem`

You can see that here (https://github.com/ucb-bar/chipyard/blob/58076cfb260a3be502d6d1c25b577da39277a7fc/generators/chipyard/src/main/scala/HarnessBinders.scala#L127-L138) there is a harness binder that connects the AXI memory port IO to the `SimAXIMem` that resides in the `TestHarness`. If you want to both instantiate a module and interact with the AXI memory port, you have to create a new aggregate mixin that combines creating the memory port with adding your new module so that you can work nicely with the harness binder / io binder setup. In other words, copy and paste together the `CanHaveMasterAXI4MemPort` and `CanHaveGCDMMIOBlackBox` into a single mixin, do the relevant connections there, then send out relevant IO's offchip.

Maybe I'm misunderstanding you though.

flp2io, 2021-02-03:
Thank you for your reply.
You got me right. 
I want to attach ModuleX, a verilog module, to the CBUS and in-between SimAXIMem and the AXI interface.
Instead of a new aggregate mixin could I just put both traits in one? Honestly, I do not know how to put them in a mixin properly.
```
trait CanHaveGCDandMasterAXI4MemPort { this: BaseSubsystem =>
//copy of CanHavePeripheryGCD
...
//copy of CanHaveMasterAXI4MemPort
...
//make connections
mem_axi4.zip(memAXI4Node.in).map { case (io, (_, edge)) =>
      gcd match {
        case Some(gcdtl) => {
          val blackbox = gcdtl.module.impl
          val s = io
// connecting axi interface coming offchip to blackbox
// AXI interface <> blackbox
          blackbox.io.s_axi_awid := s.aw.bits.id
          blackbox.io.s_axi_awaddr := s.aw.bits.addr
          blackbox.io.s_axi_awlen := s.aw.bits.len
          blackbox.io.s_axi_awsize := s.aw.bits.size
          blackbox.io.s_axi_awburst := s.aw.bits.burst
          blackbox.io.s_axi_awlock := s.aw.bits.lock 
          blackbox.io.s_axi_awcache := s.aw.bits.cache
          blackbox.io.s_axi_awprot := s.aw.bits.prot
          blackbox.io.s_axi_awqos := s.aw.bits.qos
          blackbox.io.s_axi_awvalid := s.aw.valid
          s.aw.ready  := blackbox.io.s_axi_awready
         ...
        }
        case None => None 
        }
      }
}
```
The relevant IO's for sending offchip would be the blackbox master axi ports. blackbox <> SimAXIMem
But in my understanding how diplomacy works I can not directly use them in the HarnessBinder. Maybe I missed out how to expose wires to other modules. Do I have to create an AXI bundle for the offchip wires?
And how do I access these in the HarnessBinder?

abejgonzalez, 2021-02-03:
Good to know I understand what you are trying to do. 

> Instead of a new aggregate mixin could I just put both traits in one?

I think this depends on where you want your blackbox to reside. If you want it within the SoC boundary (which I think is the "normal" approach), I think you need to create an aggregate mixin. 

*BlackBox within SoC but outside of DigitalTop:*

IIRC one of the design decisions of IOBinders/HarnessBinders was that you can only match with one lazy or non-lazy mixin (i.e. can only do something like `(system: HasPeripheryDebug)` not `(system: HasPeripheryDebug with CanHavePeripheryTLSerial)`). 

If you were able to have 1 IOBinder combine multiple mixins, I would say you could wire up your blackbox on the SoC between the `ChipTop` and `DigitalTop` layers. Something like this:

ChipTop <- AXI Out of Chip -> BlackBox <- (AXI Out of DigitalTop from `CanHaveMasterAXI4MemPort`, BB IO connections that go to CBUS) -> DigitalTop

Maybe you can do something like this by doing something like:

```
(system: CanHaveMasterAXI4MemPort) => {
  system match { 
    case systemMatched: CanHaveYourBBMixin => {
       // should have access to AXI mem port... all you need to do is connect whatever BB inputs coming from the CBUS to it
    }
```

With that being said, you would have to experiment around with this.

*BlackBox within SoC and within DigitalTop*:

With the restriction stated before with IO/HarnessBinders, the more recommended approach is to have 1 mixin create all HW within the SoC and create IO's isolated from the rest of the other IOs on the system (can't access AXI mem. interface from another mixin). If you look at the `CanHaveMasterAXI4MemPort` you can see that there is a `val mem_axi4` that automatically makes IO's off the DigitialTop module. 

https://github.com/chipsalliance/rocket-chip/blob/d71c5a674275676b9c870a20cb0a42cd039a4115/src/main/scala/subsystem/Ports.scala#L32-L65

You ideally would want to have your BB reside somewhere before that point like this (the following code is just a sketch... you have to figure out the proper names / bundles / connects / etc):

```
val axi4Bundles = memAXI4Node.out.map(_._1)
val blackBoxModule = new YourModule(....)

// connect BB to axi4 signals
blackBoxModule.io.SOMETHING := axi4Bundle.MEMBER

val newIOs = IO(new AXI4Interface)
newIOs.SOME_MEMBER := blackBoxModule.io.axi_out
```

To do this, you would have copy the `CanHaveMasterAXI4Port` splice in your code and just expose the AXI4 signals. 

*Blackbox in harness binder aka in Harness)*

This is similar to the "BlackBox within SoC but outside of DigitalTop" section. You your isolated mixin (one for your blackbox) pipe it's IO's out of the chip using an IO binder, then you have to create an aggregate HarnessBinder that matches on two mixins. You can try hacking around this in a similar manner to the aforementioned section. 

---

Does that make a bit more sense why this is non-trivial? Should be doable, just will take a little bit of hacking + experimentation.

flp2io, 2021-02-03:
Thank you very much for the details.
This is not an easy task for me.
Do I have to expose the AXI4 signals coming out of my blackbox or rather newIO in your example? How do I expose these wires in chisel?
It is not clear for me how the connection to SimAXIMem is made.

abejgonzalez, 2021-02-09:
Yup. This hacking requires a bit of knowledge about Chisel/Diplomacy to do correctly. The `axi4Bundles` should be a container of AXI4 bundles that you can access and see all of the members (size, etc). These should be wires already so then all you need to do is expose the AXI interface on your blackbox and connect things up (like what you mentioned):

```
// note... I'm not sure what the wires the bundles contain... I would look at the `AXI4Bundle` class
blackbox.io.s_axi_awid := axi4Bundles(0).aw.bits.id
blackbox.io.s_axi_awaddr := axi4Bundles(0).aw.bits.addr
// etc ...
``

At that point, I think you infer the size of the wires from the signal and pass that into the blackbox as a parameter so that its IOs can be the right size:

```
val sizeOfAXI4Bundle0AwId = axi4Bundles(0).aw.bits.id.getWidth

val blackBox = Module(new MyBlackBox(
  sizeOfAXI4Bundle0AwId,
  ....
)
```

An example of passing parameters can be seen with the CVA6 core (here: https://github.com/ucb-bar/cva6-wrapper/blob/139741a584d7e3c0446db592b5d99529bd6cf9fa/src/main/scala/CVA6CoreBlackbox.scala) (here: https://github.com/ucb-bar/cva6-wrapper/blob/139741a584d7e3c0446db592b5d99529bd6cf9fa/src/main/scala/CVA6Tile.scala#L231)

Next, you can do whatever you do with that AXI interface (compress stuff etc), then pass out those same signals and connect those signals to the output bundle:

```
// note this is similar to the link I gave below
// this should work because the AXI interface should be the same entering and exiting the BB
val ios = IO(Flipped(new HeterogeneousBag(axi4Bundles.map(_.cloneType))))
ios.suggestName("output_axi_or_some_better_name")
ios.foreach { case io =>
  io.<NAME OF MEMBER> := blackbox.io.<MEMBER COMING FROM OUTPUT OF BB>
}
```

Note this is adapted from this: https://github.com/chipsalliance/rocket-chip/blob/d71c5a674275676b9c870a20cb0a42cd039a4115/src/main/scala/diplomacy/Nodes.scala#L643-L649

This is how I would probably approach it... note that. I haven't actually done this so stuff will come up along the way (so this code is more of a sketch rather than "copy this and it will work")

closed 2021-03-15 by flp2io