ucb-bar/chipyard #764: Adding ports to SyncReadMems (foundry memories)

lsteveol, 2021-01-18:
`barstools` generates the memory wrappers for all/most `SyncReadMem`'s in the `MacroCompiler` stage of the chipyard build. This all works great however when replacing with foundry memories, there are often some additional ports on the memories. In some cases I believe these can be tied off with respective values, in other cases it's best at a minimum connect them to a control register in the event they need to be changed in silicon.

Does the current chipyard/barstools flow allow for any additional ports to be added?

I saw in the `barstools` dir (https://github.com/ucb-bar/barstools/blob/8c9387452430ab29da792d2f92891fd2fc7ccf4f/macros/src/test/scala/MacroCompilerSpec.scala#L213) the MacroCompilerSpec in the test has an `extraPorts` feature. This however seems to be related to just a test and not a part of the normal flow. Although, it seems to mostly be what I'm looking for.

I don't think this is a feature currently supported but hoping someone has dealt with it before and can correct me. Another reason I don't believe it's a feature is that barstools would need to then string the additional port all the way through the `ChipTop` hierarchy.

colinschmidt, 2021-01-18:
You can use the extra ports feature of MacroCompiler currently to tie off extra ports high or low. This has been sufficient for most of the technologies used recently with Chipyard.
Previously in a different incarnation of this flow we supported tying the ports to registers, and the current API is intended to be extended to support that functionality but it hasn't been implemented yet. It should be easier than before given all of the support FIRRTL has added for annotations that move circuit components.

You can see an explanation of the current support here:
https://github.com/ucb-bar/plsi-mdf/blob/4be9b173647c77f990a542f4eb5f69af01d77316/macro_format.yml#L132

lsteveol, 2021-01-18:
Thanks @colinschmidt for the pointer. Does barstools have a PR that I can subscribe to for checking updates? I looked through the issues in the repo but didn't see anything.

colinschmidt, 2021-01-19:
Unfortunately, I don't think there is anyone currently working on support for this, so there is no PR to track.

closed 2021-02-19 by lsteveol