ucb-bar/chipyard #768: The L1ScratchpadRocketConfig fails running rv64ui-p-add instruction

hadirkhan10, 2021-01-21:
**_chipyard release: 1.4.0_**

Hello,
I am running the following command to build the simulator:
```
make CONFIG=L1ScratchpadRocketConfig
```
then executing an `rv64ui-p-add` test on the built simulator with this command:
```
./simulator-chipyard-L1ScratchpadRocketConfig $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple
```
This results in the following error output:
```
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 45409
[UART] UART0 is here (stdin/stdout).
[5] %Error: chipyard.TestHarness.L1ScratchpadRocketConfig.top.v:141860: Assertion failed in TOP.TestHarness.chiptop.system.subsystem_l2_wrapper.InclusiveCache_inner_TLBuffer.monitor
%Error: /home/hadirkhan/Desktop/chipyard-latest/sims/verilator/generated-src/chipyard.TestHarness.L1ScratchpadRocketConfig/chipyard.TestHarness.L1ScratchpadRocketConfig.top.v:141860: Verilog $stop
Aborting...
```
How to resolve this issue?

alonamid, 2021-01-21:
I believe this was answered in the following thread:
https://groups.google.com/u/1/g/chipyard/c/fHvnRREsV5A

hadirkhan10, 2021-01-22:
No @alonamid that answer was based on the **release: 1.3.0**. This release has a different configuration for the `L1ScratchpadRocketConfig` i.e it has separate ICache and DCache configured as scratchpad memories. What is the point of adding a configuration if it does not work? Or is there another way to make the tests run on this config? I am interested in a small design with only separate scratchpad memories for instruction and data.

jerryz123, 2021-01-23:
The best minimal config is ScratchpadOnlyRocketConfig.

L1ScratchPadRocketConfig is an odd design that does not work with the default simulated bringup behavior, as the ScratchPad caches do not allow access to the backing memory, where the payload is loaded. 
In contrast, ScratchPadOnlyRocketConfig contains only scratchpad caches, and no backing memory. In that config, the payload is loaded directly into the scratchpad (since the scratchpad is assigned the physical addr of the normal DRAM base addr).

In the future, to reduce confusion, we may remove L1ScratchpadRocketConfig.

hadirkhan10, 2021-01-24:
Hi Jerry,

Is there a config or adjustment with which I can completely remove the
caching protocol and use small on-chip srams for instructions and data
storage? I am looking for a small embedded design.

QL1115, 2023-07-02:
> 

Hi, @hadirkhan10 @jerryz123 
Sorry to bother you. I wonder that did you figure out that how to completely remove the L1 ICache and DCache, I want to use scratchpad instead.
All I found on the Internet, there's only one way to configure the DCache to scratchpad...