ucb-bar/chipyard #777: SmallSPIFlashRocketConfig fails in running rv64ui-p-add test

hadirkhan10, 2021-01-28:
**_chipyard version_**: 1.4.0

Generating a simulator for the `SmallSPIFlashRocketConfig` and running an `rv64ui-p-add` using this command:
```
make CONFIG=SmallSPIFlashRocketConfig run-binary-debug BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
```
results in these errors:
```
[UART] UART0 is here (stdin/stdout).
[0] %Error: plusarg_file_mem.sv:50: Assertion failed in TOP.TestHarness.spi_mem_0.memory
%Error: /home/hadirkhan/Desktop/chipyard-latest/sims/verilator/generated-src/chipyard.TestHarness.SmallSPIFlashRocketConfig/plusarg_file_mem.sv:50: Verilog $stop
Aborting...
```

jerryz123, 2021-01-28:
If you inspect the output file `rv64ui-p-add.out`, the error message should say
```No memory image provided. Use +spiflash0=<file> to specify.```

closed 2021-01-28 by jerryz123

hadirkhan10, 2021-01-29:
@jerryz123 I am providing the `bootrom.rv64.img` file considering that now instead of a boot ROM the bootloader code is read from the Flash using the following command:
```
./simulator-chipyard-SmallSPIFlashRocketConfig +spiflash0=../../bootrom/bootrom.rv64.img $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
```
However, now I am getting the following error:
```
Listening on port 34313
[UART] UART0 is here (stdin/stdout).
Error: image file ../../bootrom/bootrom.rv64.img is not large enough to store 0x10000000 bytes.
```
Is it necessary for the image to be as large as the size of the flash? If yes then how can I reduce the Flash size to be that of the bootloader size?