ucb-bar/chipyard #780: Three build errors in IOBinders.scala.

mbs0221, 2021-02-01:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* latest, Hash: 58076cfb
*OS:* Linux inspiron-5488 5.4.0-65-generic #73-Ubuntu SMP Mon Jan 18 17:25:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
*Other:* make SUB_PROJECT=vc709 bitstream

**What is the current behavior?**
I tried to port the code for Xilinx VC709 Board from VC707. After fixing many errors of the code, I get three errors as following. When I running make SUB_PROJECT=vcu118 bitstream, It outputs the same three error.

[error] /mnt/D/Ubuntu/projects/chipyard/generators/chipyard/src/main/scala/IOBinders.scala:122:7: OverrideIOBinder is already defined as class OverrideIOBinder in package object iobinders
[error] class OverrideIOBinder[T](fn: T => IOBinderTuple)(implicit tag: ClassTag[T]) extends ConcreteIOBinder[T](false, fn)
[error]       ^
[error] /mnt/D/Ubuntu/projects/chipyard/generators/chipyard/src/main/scala/IOBinders.scala:123:7: ComposeIOBinder is already defined as class ComposeIOBinder in package object iobinders
[error] class ComposeIOBinder[T](fn: T => IOBinderTuple)(implicit tag: ClassTag[T]) extends ConcreteIOBinder[T](true, fn)
[error]       ^
[error] /mnt/D/Ubuntu/projects/chipyard/generators/chipyard/src/main/scala/IOBinders.scala:365:7: WithDontTouchPorts is already defined as class WithDontTouchPorts in package object iobinders
[error] class WithDontTouchPorts extends OverrideIOBinder({
[error]       ^
[error] three errors found
[error] (chipyard / Compile / compileIncremental) Compilation failed
[error] Total time: 41 s, completed Feb 1, 2021, 8:05:26 PM
make: *** [/mnt/D/Ubuntu/projects/chipyard/common.mk:98: generator_temp] Error 1

**What is the expected behavior?**
I ran the build command according to this instruction: [generating-a-bitstream](https://chipyard.readthedocs.io/en/latest/Prototyping/General.html#generating-a-bitstream). It is expected to run without errors.

**Other information**

I have searched the many causes of "*** is already defined as class ***". I think it may be a problem of the Scala sources configuration. But I don't know how to fix it in chipyard.

abejgonzalez, 2021-02-01:
Before you make any of your changes did things compile beforehand (from a fresh repo did the `cd fpga/ && make SUB_PROJECT=vcu118 bitstream` work)? I'm not sure what the issue is here so I want to make sure that the repository is set up properly first

mbs0221, 2021-02-02:
After removing out all the changes in `/fpga`, and deleting the `target` directories in /fpga and in /generators, I still met the same errors when running `cd fpga/ && make SUB_PROJECT=vcu118 bitstream`. Then I tried running `make verilog` both in `sims/vcs` and in `sims/verilator`. It produces the same errors as in `/fpga`.

The last line in the above records tells us a problem in `generator_temp` in `common.mk`. The main cause may be the `SCALA_SOURCES` variable.This link [Intellij compile failures: “is already defined as”](https://stackoverflow.com/questions/16885489/intellij-compile-failures-is-already-defined-as) can be a reference. But I use `VS Code` rather than 'IDEA' to build the project.

A possible solution is getting a completely fresh repository. But it is time consuming in updating all the submodules. In about half months ago, I can run `make verilog` succesfully both in `sims/verilator` and in `sims/vcs` folder. But after running `git checkout` for the updates in two days ago, it produces errors.

There are two `IOBinders.scala` in `/fpga/src/main/scala/vcu118` folder. Since they are in different packages, it may not be a cause.

But finally after removing out the `/generators/chipyard/target` folder, these errors do not appear now!

closed 2021-02-02 by mbs0221

codoor, 2022-03-28:
I met this recently. And after removing out the /generators/chipyard/target folder, these errors do not appear. But I got a another error, and I don't know how to fix it.
the error as follow:
[image]