ucb-bar/chipyard #785: Missing timing constraints

mmoghad, 2021-02-02:
Hello everyone, 

I am trying to adapt a simple Chipyard config to run on Intel Stratix 10 FPGA for prototyping.

I have looked at the new FPGA support added to the 1.4.0 release to figure out the timing constraints needed for my config (which is very similar to the default FPGA config). I could only find a few top-level constraints in the chipyard/fpga/generated-src/chipyard.fpga.vcu118[...]shell.sdc files. 

Examining the generated Verilog code however, I can see at least AsyncResetSynchronizer (which to the best of my knowledge require timing constraints), and the possibility for clock crossing that can be added to the multi clock Tilelink bus, but no related SDC. 

Am I missing something here? 
Also, is there a simple way to get rid of the multi clock stuff, and the reset synchronisers? I tried older Chipyard configs, without the WithMulticlockCoherentBusTopology fragment, but they don't seem to work anymore. 

Thank you in advance

abejgonzalez, 2021-02-09:
Hmm. Are you trying to do a multi-clock simulation on the FPGA. Last I checked I didn't need timing constraints for those crossings since I didn't use a multi-clock SoC. So to be clear... I didn't look into any timing constraints for multi-clock systems. Are you seeing multi-clock stuff when not trying to use a multi-clock config?

To add support for multi-clock on the FPGA prototypes would be super nice... but for right now that isn't a priority.

mmoghad, 2021-02-09:
I am currently using a single clock, but although the build is timing clean (at 150 MHz) it has non-deterministic behaviour on the FPGA, so I could only assume there are missing constraints. I checked FireSim and realised they use passes to get rid of some stuff (e.g. WithAsyncResetReplacement). I was wondering if we could somehow automatically generate constraints instead of using those passes?

abejgonzalez, 2021-02-09:
We could probably do some automation around this. Specifically, you could maybe add something into `barstools` (an extra pass adding a *.sdc file with the relevant constraints).

colinschmidt, 2021-02-09:
Linking a known issue where this has come up before: https://github.com/chipsalliance/chisel3/issues/1735

closed 2021-02-09 by mmoghad