ucb-bar/chipyard #788: Readthedocs 1.4: Two missing components on the Ubuntu install script

JL102, 2021-02-06:
Hello,

There are two problems with the install script provided on the readthedocs documentation:
1. `python3.6-dev`, as included in the install script, is not available on current versions of Ubuntu. See https://stackoverflow.com/questions/43621584/why-cant-i-install-python3-6-dev-on-ubuntu16-04#44735144
2. The package `autoconf` is required for installing verilator in the end of the script, but is not included in the list of items to install, nor does it come with Ubuntu 20.04. It should probably be added to the items to be installed.

Minor issues of course, but just FYI.

alonamid, 2021-02-08:
I believe ubuntu 18.04LTS and 20.04LTS have pre-installed later versions of Python, so not have python3.6 should not impact Chipyard (python3.6 is required mostly for 16.04LTS).

I find it quite surprising the `autoconf` isn't preinstalled with Ububtu, but we will happily accept a PR to add `autoconf` to the ubuntu requirements list in the docs

abejgonzalez, 2021-05-07:
Closing since this should be resolved by #874

closed 2021-05-07 by abejgonzalez