ucb-bar/chipyard #789: Can not run Chipyard on Apple Silicon (Macbook Air M1)

shi27feng, 2021-02-08:
Hi everyone,
Does anybody know how can I solve the below issue? Thanks

**Impact**: rtl | software | unknown | other

**Tell us about your environment:**
*Chipyard Version:* 1.4.0
*OS:* MAC OS Big Sur
*Other:* 

**What is the current behavior?**
I tried to run the command below and get the below results:
```shell
bash-3.2$ ./scripts/build-toolchains.sh riscv-tools
readlink: illegal option -- f
usage: readlink [-n] [file ...]
Installing toolchain to /Users/fengshi/Workspace/chipyard/riscv-tools-install
build-toolchains.sh: obsolete make version; need GNU make 4.x or later
bash-3.2$ make --version
GNU Make 4.3
Built for arm-apple-darwin20.2.0
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

**What is the expected behavior?**

**Other information**

a0u, 2021-02-08:
If found on the PATH, `gnumake` and `gmake` take precedence over `make` (to mirrror the behavior of glibc's autoconf check).  Can you confirm if those are also point to Make 4.3?

The earlier error is because BSD `readlink` does not support the same options as GNU `readlink`.  If you have GNU coreutils installed through MacPorts or Homebrew, you can replace `readlink` with `greadlink` in the `build-toolchains.sh` shell script.

Other problems will arise later in the gcc build if you're using a case-insensitive filesystem on the Mac; see the riscv-gnu-toolchain [README](https://github.com/riscv/riscv-gnu-toolchain/blob/master/README.md) for OS X prequisites.

shi27feng, 2021-02-08:
@a0u Thank you for the prompt reply, I'll try your method.

shi27feng, 2021-02-28:
I finally found the solution, just simply set $MAKE:
```shell
❯ export MAKE=make
```

closed 2021-02-28 by shi27feng