ucb-bar/chipyard #794: Does anyone have an IntelliJ setup for chipyard that allows full editing of chisel/scala code and searching code?

gabaxter, 2021-02-10:


abejgonzalez, 2021-02-15:
Most of the Chipyard devs that I know just use Vim/Emacs as their editors + Ctags/Grep to search around. If you end up setting up an IntelliJ setup, feel free to let us know what it takes to setup Chipyard that way!

ccelio, 2021-02-15:
I haven't used chipyard specifically, but I remember getting rocket-chip/riscv-boom to work (IntelliJ was a good way to navigate the complexity of rocket-chip!). The only big "trick" I remember needing to do was I found it necessary to first build rocket-chip manually, which constructed the necessary jar files, and then I told IntelliJ "Add as Library..." to point to these compiled jar files. Sorry if my information is very out-of-date or not relevant, but I do want to encourage usage of IntelliJ. It's a good tool, especially when trying to learn a new code base. =). 

There's also this blogpost which is maybe relevant : http://blog.edmondcote.com/2018/04/using-intellij-as-rocketchip-ide.html.