ucb-bar/chipyard #800: Cannot connect OpenOCD with the verilator simulator

hadirkhan10, 2021-02-22:
**Tell us about your environment:**
*Chipyard Version:* 1.4.0

**What is the current behavior?** 
I have generated a default configuration using the command below:
```
make CONFIG=TinyRocketConfig
```
Then invoking the simulator with the following command to listen for openocd connections:
```
./simulator-chipyard-TinyRocketConfig +permissive +jtag_rbb_enable=1 --rbb-port=9824 +permissive-off $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv32ui-p-bne
```
After this I am trying to connect the openocd with the following configuration file:
```
cat ./cemulator.cfg
interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 9824

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
halt
```
Then invoking the openocd with the following command:
```
openocd -f ./cemulator.cfg
```
results in the following error:
```
Open On-Chip Debugger 0.10.0+dev-01409-gb8620764c (2020-12-29-20:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:9824
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Error: fflush: Broken pipe
Error: read: count=-1, error=Broken pipe
Error: dmi_scan failed jtag scan
Error: Failed read (NOP) at 0x12; value=0xffffffff, status=2
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: fflush: Broken pipe
Error: failed: -4
```
**What is the expected behavior?**
The openocd should connect with the verilator simulator

**Other information**
The config used for the Arty FPGA support also uses the `TinyRocketConfig` so the generated RTL should have a proper debug module implemented. Not sure what is causing this issue.

hadirkhan10, 2021-03-02:
@jamesdunn @jerryz123 anyone looking into this problem?

jamesdunn, 2021-03-08:
It sounds like your issue is identical to [this one](https://github.com/riscv/riscv-openocd/issues/457).

In that issue, the user found that it was an OS configuration causing issues with their OpenOCD build.

I also noticed that your issue here in sim is similar to what you experienced [here](https://github.com/riscv/riscv-openocd/issues/581) in hardware (error related to flush).

Could you give us some information about your OS configuration?

hadirkhan10, 2021-03-09:
No I built it on the same OS, in-fact I also downloaded the already prebuilt binaries of OpenOCD and RISCV GNU Toolchain from the SiFive website.

My Ubuntu version is: Ubuntu 20.04.1 LTS

jamesdunn, 2021-03-16:
I was able to reproduce this error on both Ubuntu 20.04 and Centos 7, as well as with different Rocket configurations (TinyRocketConfig and RocketConfig).

For future debugging, please run OpenOCD with the verbose `-d` flag and the verilator binary with the `+verbose` option.

Similar to the issue I mentioned in my first comment, I see a Signal 13 termination in the OpenOCD output, which means that a pipe has been closed prematurely, probably by the simulator.

When I run the simulator in verbose mode, I see that the it actually passes after 18125 cycles.
```
*** PASSED *** Completed after 18125 cycles
```

Do you see the same behavior?

The verbose verilator output gives the full assembly trace, so I'll need to look into that to see what's going on.

jerryhethatday, 2022-12-19:
Hey @jamesdunn I recently met a bizarre issue when tring to install the riscv openocd , I post the issue in chipyard maillist: https://groups.google.com/g/chipyard/c/82qT9DzWtuA .
Do you mind taking a look ? Thanks