ucb-bar/chipyard #804: Push ASAP7 SRAM Macros into the P&R Flow

baichen318, 2021-02-24:
Hi all,

I have used ASAP7 SRAM macros (https://github.com/ucb-bar/hammer/tree/master/src/hammer-vlsi/technology/asap7) to synthesize a BOOM SoC. (I can use the synthesized BOOM SoC to simulate with different benchmarks)

I use Genus 18.13-s027_1 to get `ChipTop.mapped.v`. However, when I use Innovus v17.11-s080_1 and push `ChipTop.mapped.v` to the P&R flow, I am failed and I get an error prompted by Innovus and it says `ERROR (NRIG-92) INSTANCE system/boom_tile/dcache/data/array_0_0/array_0_0_ext/mem_0_1 is not placed on the manufacturing grid. All instances must be legally placed. Correct placement before continuing.` (In fact, mem_0_1 is SRAM2RW128x4)

I was wondering how I can push `ChipTop.mapped.v` together with `sram_behav_models.v` to the P&R flow?

Thank you very much!

Best regards,
Chen

harrisonliew, 2021-02-24:
Hi Chen,

It looks like you have not actually placed the generated macros into your floorplan using `vlsi.inputs.placement_constraints`.

baichen318, 2021-02-25:
> Hi Chen,
> 
> It looks like you have not actually placed the generated macros into your floorplan using `vlsi.inputs.placement_constraints`.

Hi Harrison,

Thank you for your kind reply! I am going to fix it.

Best regards,
Chen

closed 2021-03-07 by alonamid

baichen318, 2021-07-20:
Hi all,

I have placed the generated SRAM macros into my floorplan using `vlsi.inputs.placement_constraints`, and the generated floorplan.tcl is attached as shown in Fig.
[image]
Nevertheless, Innovus outputs an error when it executes the command: `edit_pin -fixed_pin -pin * -hinst ChipTop -pattern fill_optimised -layer { M5 M7 } -side bottom -start { 3000 0 } -end { 0 0 }` and the error is shown below .
[image]
The error says `Layer M10: Pitch=11520x9 is still less than min width=32000 + min spacing=640.`
In the command `ccopt_design -hold -report_dir hammer_cts_debug -report_prefix hammer_cts`, Innovus outputs a induced error, `The route/traversal graph for net 'CTS_47' is not fully connected.` and terminates.

I was wondering how I can solve this problem.

Thank you all!

Best regards,
Chen