ucb-bar/chipyard #814: Setup script bug, version incompatibility with Bison 3.7

allisonrandal, 2021-03-04:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* current master
*OS:* Linux muon 5.8.0-7630-generic #32~1606339263~20.10~61c3910-Ubuntu SMP Thu Nov 26 00:10:35 UTC  x86_64 x86_64 x86_64 GNU/Linux

*Other:* Following the documentation at https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html, the instructions for Ubuntu/Debian (which are included in the docs from scripts/ubuntu-req.sh) instruct the user to to clone the Verilator repo, and then run:

git checkout v4.034

There is a known bug in that version of the verilator source code with Bison 3.7 (which ships with more recent versions of Debian/Ubuntu), as reported at:

https://www.mail-archive.com/debian-bugs-rc@lists.debian.org/msg569900.html

The bug has been fixed in more recent versions of the verilator source code:

https://www.mail-archive.com/debian-bugs-rc@lists.debian.org/msg571592.html

**What is the current behavior?**

Verilator fails to compile with the error message:

verilog.c:313:10: fatal error: verilog.h: No such file or directory

**What is the expected behavior?**

Verilator should compile.

**Other information**

If you skip the instruction to checkout v4.034, it all works fine, but if you follow that step Verilator won't compile on more recent versions of of Debian/Ubuntu. Simply removing that line from ubuntu-req.sh is probably an adequate fix.

jerryz123, 2021-03-04:
Thanks for the report. It looks like we should just bump to Verilator v4.100

jerryz123, 2023-02-17:
Done

closed 2023-02-17 by jerryz123