ucb-bar/chipyard #816: How do I compile a c program such that the binary can be run/simulated by an RTL config?

JL102, 2021-03-05:
Hello,

I am attempting to use `riscv64-unknown-elf-gcc` to compile a simple test program, and have it run on an RTL config (from section 2.1 in the docs). 

This is the test file, `hello.c`. I initially had it print a "hello world" statement, but changed it such that it does not need any imports or libraries:
```c
int main() {
   int x = 5;
   return 0;
}
```
This is what I am doing:
- To compile: `riscv64-unknown-elf-gcc -g hello.c -o hello-riscv`
 - I am able to simulate it with Spike successfully: `spike pk hello-riscv` runs without errors. (When my "hello world" program had a printf statement, it printed correctly as well.)
 - Then, I attempt to run the executable with an RTL config. Inside `chipyard/sims/verilator`, I run: `./simulator-chipyard-RocketConfig ~/hello-riscv`
This is the output I get: 
```
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36041
[UART] UART0 is here (stdin/stdout).
warning: tohost and fromhost symbols not in ELF; can't communicate with target
```
and then it hangs indefinitely. Upon examining `top`, I can see that `simulator-chipy` continues to utilize 100% of a CPU thread.

Is there anything I'm doing wrong? How do I compile the program such that it runs correctly? I could not find any information about tohost and fromhost, aside from chapter 6.12, which talks about booting a Linux kernel. I'm not attempting to boot Linux via the RISC-V cpu; I just want it to run a certain binary like in the default examples.

alonamid, 2021-03-07:
Could you try running `make run-binary CONFIG=RocketConfig BINARY=/full/path/hello-riscv` instead of `./simulator-chipyard-RocketConfig ~/hello-riscv` ?

JL102, 2021-03-08:
Just did now. Same thing, except I don't get the "Listening on port", "UART0", and "warning: tohost and fromhost" messages.

```bash
drak@JBOOK-PRIMARY:~/chipyard/sims/verilator$ make run-binary CONFIG=RocketConfig BINARY=/home/drak/hello-riscv
Running with RISCV=/home/drak/chipyard/esp-tools-install
(set -o pipefail &&  /home/drak/chipyard/sims/verilator/simulator-chipyard-RocketConfig +permissive +dramsim +dramsim_ini_dir=/home/drak/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000   +verbose +permissive-off /home/drak/hello-riscv </dev/null 2> >(spike-dasm > /home/drak/chipyard/sims/verilator/output/chipyard.TestHarness.RocketConfig/hello-riscv.out) | tee /home/drak/chipyard/sims/verilator/output/chipyard.TestHarness.RocketConfig/hello-riscv.log)
```
The process hangs and continues to utilize 100% of the CPU core, as before.

If I attempt to pass it a file that does not exist, or if I pass it an executable that was compiled for x64, the process ends with error 255.

lxu28973, 2021-03-14:
I encountered the same problem. I compile my c program with riscv64-unknown-elf-gcc, and get the ELF file a.out. I run ./simulator-chipyard-MegaBoomConfig pk ~/H/hello-world/Gslexample/a.out, and it just hangs. After some search, I try ./simulator-chipyard-MegaBoomConfig pk ~/H/hello-world/Gslexample/a.out +dramsim. It seems like stuck, but after a long time(about 30 min), it gets the results. So, maybe just be patient.

alonamid, 2021-03-31:
@JL102 , I noticed that when you ran your program on Spike, you ran it with pk. Was it able to run on Spike without pk as well?

@RaoShri , It would be useful to run the first make command (`make run-binary`) in verbose mode, so we can see where it actually fails. Regarding calling the simulator directly (your second method), have you tried running the same test program with RocketConfig? That would be a good sanity check to confirm whether the issue is with running the program, or rather with some internal modification in your CustomConfig

RaoShri, 2021-03-31:
@alonamid ,
Yes I get the same result when using second method with RocketConfig as well :

└─$ ./simulator-chipyard-RocketConfig $RISCV/bin/target/share/riscv-tests/isa/rv64ui-p-add                                                                                                                                             
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 46735
[UART] UART0 is here (stdin/stdout).

About adding the verbose to the first command, how do I do it? Enabling verbose would help me debug a lot of things faster. Might seem like a trivial question, bear with me, I am new to this.

RaoShri, 2021-04-01:
@alonamid When running with run-make-binary, I was able to see the errors and fix it.

└─$ make run-binary-debug BINARY=$RISCV/bin/target/share/riscv-tests/isa/rv64ui-p-add CONFIG=ShrinidhiCustomConfig
Running with RISCV=/opt/riscv/bin
(set -o pipefail &&  /home/shrinidhi/Project/chipyard/sims/verilator/simulator-chipyard-ShrinidhiCustomConfig-debug +permissive +dramsim +dramsim_ini_dir=/home/shrinidhi/Project/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000   +verbose -v/home/shrinidhi/Project/chipyard/sims/verilator/output/chipyard.TestHarness.ShrinidhiCustomConfig/rv64ui-p-add.vcd +permissive-off /opt/riscv/bin/bin/target/share/riscv-tests/isa/rv64ui-p-add </dev/null 2> >(spike-dasm > /home/shrinidhi/Project/chipyard/sims/verilator/output/chipyard.TestHarness.ShrinidhiCustomConfig/rv64ui-p-add.out) | tee /home/shrinidhi/Project/chipyard/sims/verilator/output/chipyard.TestHarness.ShrinidhiCustomConfig/rv64ui-p-add.log)
[UART] UART0 is here (stdin/stdout).
         
         
Again terminal print is blank, but I see the test has passed in the log. 
How do I add verbose to the below command?

└─$ ./simulator-chipyard-ShrinidhiCustomConfig $RISCV/bin/target/share/riscv-tests/isa/rv64ui-p-bltu 
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 46485
[UART] UART0 is here (stdin/stdout).
   

I see this, but no corresponding log/out file in output/chipyard.TestHarness.ShrinidhiCustomConfig. I am guessing this test also has passed but I have no print in terminal nor any log file for it.

lpppeipei, 2021-12-24:
> t +dramsim

Hi  RaoShri,

Have  you saved your problem,I  got  the  similar question with you.

liupeipei@ubuntu:~/chipyard/sims/verilator$ ./simulator-chipyard-RocketConfig $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 38123
[UART] UART0 is here (stdin/stdout).
liupeipei@ubuntu:~/chipyard/sims/verilator$ make run-binary BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple
Running with RISCV=/home/liupeipei/chipyard/riscv-tools-install
(set -o pipefail &&  /home/liupeipei/chipyard/sims/verilator/simulator-chipyard-RocketConfig +permissive +dramsim +dramsim_ini_dir=/home/liupeipei/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000   +verbose +permissive-off /home/liupeipei/chipyard/riscv-tools-install/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple </dev/null 2> >(spike-dasm > /home/liupeipei/chipyard/sims/verilator/output/chipyard.TestHarness.RocketConfig/rv64ui-p-simple.out) | tee /home/liupeipei/chipyard/sims/verilator/output/chipyard.TestHarness.RocketConfig/rv64ui-p-simple.log)
[UART] UART0 is here (stdin/stdout).
liupeipei@ubuntu:~/chipyard/sims/verilator$ make CONFIG=GCDTLRocketConfig BINARY=../../tests/gcd.riscv run-binary
Running with RISCV=/home/liupeipei/chipyard/riscv-tools-install
(set -o pipefail &&  /home/liupeipei/chipyard/sims/verilator/simulator-chipyard-GCDTLRocketConfig +permissive +dramsim +dramsim_ini_dir=/home/liupeipei/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000   +verbose +permissive-off ../../tests/gcd.riscv </dev/null 2> >(spike-dasm > /home/liupeipei/chipyard/sims/verilator/output/chipyard.TestHarness.GCDTLRocketConfig/gcd.out) | tee /home/liupeipei/chipyard/sims/verilator/output/chipyard.TestHarness.GCDTLRocketConfig/gcd.log)
[UART] UART0 is here (stdin/stdout).
make: *** [/home/liupeipei/chipyard/common.mk:175: run-binary] Error 255

lpppeipei, 2021-12-24:
> Hi, I tried to run the binary from pre-built riscv-tests, using the following command: make run-binary BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple
> 
> I am getting the below error:
> 
> [UART] UART0 is here (stdin/stdout). make: *** [/home/shrinidhi/Project/chipyard/common.mk:162: run-binary] Error 255
> 
> When I try to run tests this way: └─$ ./simulator-chipyard-BoomCustomConfig $RISCV/bin/target/share/riscv-tests/isa/rv64ui-p-add
> 
> I get the UART0 print but nothing after that: This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1. Listening on port 34083 [UART] UART0 is here (stdin/stdout).
> 
> Am I missing anything here? Please help out.

I  got the same question  with you , have  you  saved your problem ? And  how ?   Thanks!

scottzhang, 2022-01-16:
Please read the document carefully. 

https://chipyard.readthedocs.io/en/latest/Software/Baremetal.html

#include <stdio.h>

int main(void)
{
    printf("Hello, World!\n");
    return 0;
}

$ riscv64-unknown-elf-gcc -fno-common -fno-builtin-printf -specs=htif_nano.specs -c hello.c
$ riscv64-unknown-elf-gcc -static -specs=htif_nano.specs hello.o -o hello.riscv
$ spike hello.riscv
Hello, World!

root@51df727fcb26:~/chipyard/sims/verilator# make run-binary BINARY=./hello.riscv
Running with RISCV=/root/chipyard/esp-tools-install
(set -o pipefail &&  /root/chipyard/sims/verilator/simulator-chipyard-RocketConfig +permissive +dramsim +dramsim_ini_dir=/root/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000   +verbose +permissive-off ./hello.riscv </dev/null 2> >(spike-dasm > /root/chipyard/sims/verilator/output/chipyard.TestHarness.RocketConfig/hello.out) | tee /root/chipyard/sims/verilator/output/chipyard.TestHarness.RocketConfig/hello.log)
[UART] UART0 is here (stdin/stdout).
Hello, World!
root@51df727fcb26:~/chipyard/sims/verilator#

JL102, 2022-01-17:
Thanks, I hadn't even seen that page myself. Probably because the title ("Baremetal RISC-V Programs") didn't stick out to me as what I had been looking for.

The way I had solved the problem was by copying the contents of the `riscv-tests/benchmarks` Makefile, as well as the `env` folder for includes - https://github.com/ucb-bar/esp-tests/blob/7c3dfd3205fa90104db06346e797a8eeea149207/benchmarks/Makefile

All the command-line arguments that they use is black magic to me, but the resulting executable takes even less time to simulate.

Simulating executable with the riscv-tests black magic:
```drak@JDESK:~/chipyard/sims/verilator$ ./simulator-chipyard-RocketAndCustomConfig-debug -c ~/chipyard/generators/customtensor/benchmarks/temp.riscv
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 33163
[UART] UART0 is here (stdin/stdout).
Hello, World!
*** PASSED *** Completed after 90774 cycles
```
Simulating executable with the suggested command line arguments:
```
drak@JDESK:~/chipyard/sims/verilator$ ./simulator-chipyard-RocketAndCustomConfig-debug -c ~/chipyard/generators/customtensor/benchmarks/temp/hello.riscv
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 41369
[UART] UART0 is here (stdin/stdout).
Hello, World!
*** PASSED *** Completed after 117518 cycles
```

Here's my benchmarks folder with that modified Makefile: https://github.com/JL102/rowan-riscv-tensor/tree/main/benchmarks

scottzhang, 2022-01-17:
well. Without linux or any other os is Bare Metal. And chipyard doesn't ship with an booting/running linux, it is baremetal.

JL102, 2022-02-02:
Closing as the question has been answered. (Thanks again)

closed 2022-02-02 by JL102

mudassar50, 2024-10-03:
(/home/u20/chipyard/.conda-env) u20@u20-VirtualBox:~/chipyard/sims/verilator$ make CONFIG=AdderTLRocketConfig BINARY=../../tests/adder.riscv run-binary VERBOSE=1
Running with RISCV=/home/u20/chipyard/.conda-env/riscv-tools
(set -o pipefail &&  /home/u20/chipyard/sims/verilator/simulator-chipyard-AdderTLRocketConfig +permissive +dramsim +dramsim_ini_dir=/home/u20/chipyard/generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000   +verbose +permissive-off ../../tests/adder.riscv </dev/null 2> >(spike-dasm > /home/u20/chipyard/sims/verilator/output/chipyard.TestHarness.AdderTLRocketConfig/adder.out) | tee /home/u20/chipyard/sims/verilator/output/chipyard.TestHarness.AdderTLRocketConfig/adder.log)
[UART] UART0 is here (stdin/stdout).
Enter ADDER computation:
make: *** [/home/u20/chipyard/common.mk:206: run-binary] Error 7
(/home/u20/chipyard/.conda-env) u20@u20-VirtualBox:~/chipyard/sims/verilator$ 

how can i solve this problem