ucb-bar/chipyard #818: Update version of Spike

allisonrandal, 2021-03-05:
**Impact**: software

**Description**

Chipyard currently checks out the Spike repo at riscv/riscv-isa-sim@acd953afd2f52d64e2264c2c7c713dc0ad614406 from March 2020, however this version of Spike fails to compile on recent versions of Ubuntu/Debian because of missing import lines, which were fixed in May 2020:

https://github.com/riscv/riscv-isa-sim/commit/b3855682c2d744c613d2ffd6b53e3f021ecea4f3#diff-490b36666b1c906a8b80b8c0166084b2f4d99b555ee9614434d580b14a9f74c4

I've worked around the problem locally by adding the line:

  `#include <stdexcept>`

to the files fesvr/dtm.cc and riscv/devices.h in toolchains/riscv-tools/riscv-isa-sim. This patch allows Spike to compile successfully.

**What is a motivating example for changing the behavior?**

Spike should compile without manually patching the source.

alonamid, 2021-03-07:
We've recently built the version of Spike that submoduled in chipyard on the latest Debian stable and did not observe any problems. Could you give us some more details about the version of Debian you are running on, and what errors you were getting?

allisonrandal, 2021-03-08:
The current Debian stable version is "buster", first released in July 2019. It ships GCC version 8.3. I am running Ubuntu 20.10, which ships GCC version 10.2. The same fix for a different file in Spike was specifically attributed to GCC 10.2 on Ubuntu 20.10:

https://github.com/riscv/riscv-isa-sim/pull/603

The error is repeated variants of this, on different lines of the two files:

`../fesvr/dtm.cc:488:16: error: ‘runtime_error’ is not a member of ‘std’`

It is caused by calling `std::runtime_error`, without first including `stdexcept`, which is not a problem on earlier versions of GCC.

It's worth noting that in the very latest version of Spike, the file `riscv/devices.h` has removed all calls to `std::runtime_error`, while `fesvr/dtm.cc` added the include line for `stdexcept`.

allisonrandal, 2021-03-08:
It is, BTW, totally fine to for Chipyard to decide to only support Debian stable and Ubuntu 20.04 LTS releases, and not the more recent Ubuntu 20.10 (which is a supported release, but not a long-term support release). If that is your decision, you need to be clear about it in the install guide.

And, you will need to address this issue at some point, since current Debian testing (bullseye) also ships GCC 10.2. But, presumably you'll update versions on a lot of dependencies when you upgrade to the next Debian stable, so delaying the update of Spike until that point might make a lot of sense.

alonamid, 2021-06-15:
We have bumped spike in release 1.5.0

allisonrandal, 2021-06-17:
Great, thanks!

closed 2021-06-30 by jerryz123