ucb-bar/chipyard #819: Chipyard installation fails

CuriousSingularity, 2021-03-06:
**Impact**: rtl | software | unknown | other
Software

**Tell us about your environment:**
*Chipyard Version:* 58076cfb260a3be502d6d1c25b577da39277a7fc
*OS:* `Linux Windows-Desktop-Dell 4.4.0-19041-Microsoft #488-Microsoft Mon Sep 01 13:43:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux`

*Other:* 
Setup steps from the link:
https://chipyard.readthedocs.io/en/dev/Chipyard-Basics/Initial-Repo-Setup.html

After the necessary dependencies installation, following commands were issued:
```
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
./scripts/init-submodules-no-riscv-tools.sh
```
The error stated below is encounter when the following command is executed:
```
./scripts/build-toolchains.sh riscv-tools
```

**What is the current behavior?**
```
cannot stat '/mnt/d/Data/Technical/Repositories/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/dlfcn/stamp.oST': No such file or directory
make[3]: *** [/mnt/d/Data/Technical/Repositories/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/dlfcn/stamp.oS] Error 1
../o-iterator.mk:9: recipe for target '/mnt/d/Data/Technical/Repositories/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/dlfcn/stamp.oS' failed
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/mnt/d/Data/Technical/Repositories/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-glibc/dlfcn'
make[2]: *** [dlfcn/subdir_lib] Error 2
Makefile:258: recipe for target 'dlfcn/subdir_lib' failed
make[2]: Leaving directory '/mnt/d/Data/Technical/Repositories/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/mnt/d/Data/Technical/Repositories/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d'
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2
Makefile:234: recipe for target 'stamps/build-glibc-linux-rv64imafdc-lp64d' failed
```

**What is the expected behavior?**
successful installation

**Other information**

May I know what is causing the issue? Any support would be helpful.

tymcauley, 2021-03-06:
Hmm, this looks like it could be this issue: https://github.com/riscv/riscv-gnu-toolchain#troubleshooting-build-problems

> If building a linux toolchain on a MacOS system, or on a Windows system using the Linux subsystem or cygwin, you must ensure that the filesystem is case-sensitive. A build on a case-insensitive filesystem will fail when building glibc because *.os and *.oS files will clobber each other during the build eventually resulting in confusing link errors.

Here is the PR that added that note: https://github.com/riscv/riscv-gnu-toolchain/pull/529

It has some links to related issues, so maybe there will be some tips in there for how to resolve the issue.

hust-gaoyujing, 2022-04-28:
Hi,all.I wonder if this problem is solved,and how?

michael-etzkorn, 2022-04-28:
If you build entirely within WSL the build tool script works (barring other environmental shenanigans). The fact @CuriousSingularity is building on `/mnt` here suggests they're sharing across to Windows where the filesystem is case-insensitive which, as tymcauley has already pointed out, is problematic for the build tool.

SanTaiZi-NeZha, 2022-08-01:
> Hmm, this looks like it could be this issue: https://github.com/riscv/riscv-gnu-toolchain#troubleshooting-build-problems
> 
> > If building a linux toolchain on a MacOS system, or on a Windows system using the Linux subsystem or cygwin, you must ensure that the filesystem is case-sensitive. A build on a case-insensitive filesystem will fail when building glibc because *.os and *.oS files will clobber each other during the build eventually resulting in confusing link errors.
> 
> Here is the PR that added that note: [riscv-collab/riscv-gnu-toolchain#529](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/529)
> 
> It has some links to related issues, so maybe there will be some tips in there for how to resolve the issue.

I am using msys2, how can I set case sensitivity?

jerryz123, 2023-02-17:
We only support Linux officially, and the conda-build flow should prevent issues like this in the future.

closed 2023-02-17 by jerryz123