ucb-bar/chipyard #823: Clock and Reset IOCells

KPrabs106, 2021-03-11:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* 1.4.0

**What is the current behavior?**
IO Cells generated for clock and reset do not use IOCellKey, resulting in GenericDigitalInIOCell

**What is the expected behavior?**
Clock and reset IO cell generation should be using IOCellKey

**Other information**

I might have misunderstood how to use IOCellKey, but I think the following changes are needed:

https://github.com/ucb-bar/chipyard/blob/e3d23f10345cfafd9a2f4f8aacfe07c2746699f9/generators/chipyard/src/main/scala/Clocks.scala#L28

should be
`   
 val (reset_io, resetIOCell) = IOCell.generateIOFromSignal(async_reset_wire, "reset", p(IOCellKey),
`
and 
https://github.com/ucb-bar/chipyard/blob/e3d23f10345cfafd9a2f4f8aacfe07c2746699f9/generators/chipyard/src/main/scala/Clocks.scala#L97
should be
`
      val (clock_io, clockIOCell) = IOCell.generateIOFromSignal(clock_wire, "clock", p(IOCellKey))
`

jerryz123, 2021-03-11:
Thanks for catching this. Can you open a PR against dev to fix it?

jerryz123, 2021-03-11:
Fixed, thanks for the PR

PR #824 (2021-03-11): Fix IOCell generation for clock and reset to use IOCellKey
**Related issue**: 
Fixes #823 

**Type of change**: bug fix 

**Impact**: rtl change

**Release Notes**

Use IOCellKey to generate IO Cells for clock and reset
closed 2021-03-11 by jerryz123