ucb-bar/chipyard #825: Help firmware uploading

santos-lucasm, 2021-03-11:
I was working directly with the Freedom repository with Arty35T, but I'm changing to Chipyard.
Already generated a bitstream and firmware using Chipyard.

Now I have the binaries to flash to the board, but I was using the freedom-e-sdk makefile to upload applications to the board, which seems to not work anymore with this new generated firmware.
How should I upload applications to the board using Chipyard generated firmware?

hadirkhan10, 2021-03-12:
@santos-lucasm are you using the fpga prototyping flow to generate your design for the arty board? You might be only able to use the JTAG to program the SoC and doing that is difficult. By difficult it means lots of errors which doesn't make sense. I am already stuck in this loop as well. What Freedom allows you to do is that it creates a Boot ROM which jumps to the SPI Flash base address. That is the path where you can initially load your own firmware and stop the bootloader code from jumping to the SPI Flash. Obviously, you will be limited to the size that is available but it could well help you to at least verify the RTL. Chipyard for now does not support generating a Boot ROM, I have already opened an issue [here](https://github.com/ucb-bar/chipyard/issues/813#issue-821751537) and @jamesdunn might be looking at it.

jamesdunn, 2021-03-22:
Hello @santos-lucasm—the branch `arty-spi-flash` should now allow you to upload a program to the Arty's flash memory using the freedom-e-sdk. More information can be found [here](https://github.com/ucb-bar/chipyard/issues/813#issue-821751537).

closed 2021-03-22 by jamesdunn

JACKLIAO0, 2022-01-03:
have you resolved the problem?

I have the same question why upload a binary program via free-e-sdk:

Error: Debug Module did not become active. dmcontrol=0x0
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

hope for your reply,thanks