ucb-bar/chipyard #827: Question about compile program for RocketTinyCore

LIUQyou, 2021-03-12:
Hi!
I have a question about compiling programs for 32-bit cores. I tried to use Verilator to test Bare Metal RISC-V Programs with TinyCore( Is it 32-bit core, right?), but I am not sure what kind of instruction should I use. I take the reference like the following, but it does not work.
$ riscv32-unknown-elf-gcc -fno-common -fno-builtin-printf -specs=htif_nano.specs -c hello.c
$ riscv32-unknown-elf-gcc -static -specs=htif_nano.specs hello.o -o hello.riscv

It shows like the following,
/home/liu/rocket-tools/lib/gcc/riscv32-unknown-elf/7.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lgloss_htif
collect2: error: ld returned 1 exit status
cp: cannot stat 'blackbox2.riscv': No such file or directory

Could you tell me what should I do to compile my program? My goal is to use .riscv file like the following.
./simulator-chipyard-RoccExampleTinyConfig hello.riscv

Thanks in advance,
Best regards,
Liu

jerryz123, 2021-03-24:
Was your program able to run in spike?

Yuxin-Yu, 2022-10-20:
Hello, have you solved is.I met the same problem. In the sim/inverter directory, I use the "riscv32-unknown-elf - *" toolchain to compile TinyRocketConfig, and then run the command
./simulator chipard TinyRocketConfig $RISCV/riscv32 unknown elf/share/riscv tests/benchmark/chrystone.riscv
, but I encountered an error，which like:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36189
[UART] UART0 is here (stdin/stdout).
[108000] %Error: chipyard.TestHarness.TinyRocketConfig.harness.v:25587: Assertion failed in TOP.TestHarness.ram.buffer_1.monitor
%Error: /home/yyx/riscv/projects/chipyard/sims/verilator/generated-src/chipyard.TestHarness.TinyRocketConfig/chipyard.TestHarness.TinyRocketConfig.harness.v:25587: Verilog $stop
Aborting...`
. Then I also increased the size of dtim and icache in TinyRocketConfig to 32KB. I ran again and encountered an error:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36987
[UART] UART0 is here (stdin/stdout).
*** FAILED *** (tohost = 1337)
*** FAILED *** (code = 1337, seed 1666237832) after 117165 cycles`
How can I fixed this? Thank you.

Yuxin-Yu, 2022-10-20:
Hello, have you solved is.I met the same problem. In the sim/inverter directory, I use the "riscv32-unknown-elf - *" toolchain to compile TinyRocketConfig, and then run the command
`./simulator` chipard TinyRocketConfig $RISCV/riscv32-unknown-elf/share/riscv-tests/benchmark/chrystone.riscv `
, but I encountered an error，which like:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1. Listening on port 36189 [UART] UART0 is here (stdin/stdout). [108000] %Error: chipyard.TestHarness.TinyRocketConfig.harness.v:25587: Assertion failed in TOP.TestHarness.ram.buffer_1.monitor %Error: /home/yyx/riscv/projects/chipyard/sims/verilator/generated-src/chipyard.TestHarness.TinyRocketConfig/chipyard.TestHarness.TinyRocketConfig.harness.v:25587: Verilog $stop Aborting...`
. Then I also increased the size of dtim and icache in TinyRocketConfig to 32KB. I ran again and encountered an error:
`This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1. Listening on port 36987 [UART] UART0 is here (stdin/stdout). *** FAILED *** (tohost = 1337) *** FAILED *** (code = 1337, seed 1666237832) after 117165 cycles`
How can I fixed this? Thank you.

Yuxin-Yu, 2022-10-20:
Hello, have you solved is.I met the same problem. In the sim/inverter directory, I use the "riscv32-unknown-elf - *" toolchain to compile TinyRocketConfig, and then run the command

> ./simulator chipard TinyRocketConfig $RISCV/riscv32 unknown elf/share/riscv tests/benchmark/dhrystone.riscv
 but I encountered an error，which like:

> This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.

> Listening on port 36189
> [UART] UART0 is here (stdin/stdout).
> [108000] %Error: chipyard.TestHarness.TinyRocketConfig.harness.v:25587: Assertion failed in TOP.TestHarness.ram.buffer_1.monitor
> %Error: [...]

Yuxin-Yu, 2022-10-20:
Hello, have you solved is.I met the same problem. In the sim/inverter directory, I use the "riscv32-unknown-elf - *" toolchain to compile TinyRocketConfig, and then run the command

> ./simulator chipard TinyRocketConfig $RISCV/riscv32 unknown elf/share/riscv-tests/benchmark/chrystone.riscv

but I encountered an error，which like:

> This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1. Listening on port 36189 [UART] UART0 is here (stdin/stdout). [108000] %Error: [...]

Then I also increased the size of dtim and icache in TinyRocketConfig to 32KB. I ran again and encountered an error:

> This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1. Listening on port 36987 [UART] UART0 is here (stdin/stdout). *** FAILED *** (tohost = 1337) *** FAILED *** (code = 1337, seed 1666237832) after 117165 cycles

How can I fixed this? Thank you.

michael-etzkorn, 2022-10-20:
Unfortunately, the TinyRocketCore is [not well supported at the user level](https://github.com/ucb-bar/chipyard/issues/1188) in Chipyard. Even though your DTIM is larger, you may need to adjust the linker script since it's possible your binary is still accessing memory it thinks it has but really doesn't. 

Full disclosure: I have no idea what the failure codes mean and would have to dig around in the code to find where they come from and what they signify.

Yuxin-Yu, 2022-10-21:
OK.Thank you.

Yuxin-Yu, 2022-10-21:
> > but I encountered an error
> 
> Unfortunately, the TinyRocketCore is [not well supported at the user level](https://github.com/ucb-bar/chipyard/issues/1188) in Chipyard. Even though your DTIM is larger, you may need to adjust the linker script since it's possible your binary is still accessing memory it thinks it has but really doesn't.
> 
> Full disclosure: I have no idea what the fesvr failure codes mean and would have to dig around in the code to find where they come from and what they signify.
> 
> > How can I fixed this? Thank you.
> 
> I'd migrate this discussion to #1188

Again, I used the rv32 tool chain to run RV32RocketConfig, and also successfully ran dhrystone.riscv under the rv32 tool chain. Both are rv32 structured. What's the difference between them? Why does TinyRocketConfig have problems.
@michael-etzkorn

michael-etzkorn, 2022-10-21:
I’d first confirm the linker script matches TinyConfig’s uarch. Especially, after hearing you were able to get farther by increasing the dtim size