ucb-bar/chipyard #830: Error building FireSim in the Chipyard

derAlteBarbarossa, 2021-03-18:
**Impact**: software
**Environment:**
*Chipyard Version:* Latest version, Hash: e3d23f10345cfafd9a2f4f8aacfe07c2746699f9
*OS:*`Centos7 3.10.0-957.5.1.el7.x86_64 x86_64 GNU/Linux`
I am trying it on one of the EC2 instances (As instructed by FireSim)

When I run `./scripts/firesim-setup.sh --fast`, it runs to the following error:
`error: Failed build dependencies:`
`	openssl is needed by kernel-3.10.0-957.5.1.el7.x86_64`

In spite of having openssl installed:
`~/chipyard$ openssl version`
`OpenSSL 1.1.1  11 Sep 2018`

It's also worth of mentioning that on the same machine, I could install FireSim. But when trying to install it from Chipyard, it runs to the aforementioned issue. 

Thanks for your help in advance

alonamid, 2021-03-19:
At first glance, this doesn't look like a Chipyard-related failure. It looks like an OS distribution package manager error.

Can you tell us a little more regarding what you mean by "on the same machine, I could install FireSim"?
I'm slightly confused, since the `./scripts/firesim-setup.sh --fast` script simply [calls the FireSim installation script](https://github.com/ucb-bar/chipyard/blob/master/scripts/firesim-setup.sh).
FireSim and Chipyard are released at the same time, so the FireSim commit within Chipyard should be synced.

derAlteBarbarossa, 2021-03-19:
I managed to instal FireSim on an FPGA Developer AMI of Amazon. I could launch it and simply run some workloads. I saved the image with this installed and working FireSim. Later, when I try to follow Chipyard's instruction, this error occurs. Is there any way, to link the already installed version of FireSim to Chipyard?

alonamid, 2021-03-23:
Chipyard and FireSim currently support a hybrid structure, in which either of them can act as the "top level". If you look at the `firesim/target-design` directory within your firesim repo, you will find Chipyard submoduled there (with the firesim submodule within Chipyard disabled).

However, I still find this error to be pretty strange. Are you running `./scripts/firesim-setup.sh --fast` after running the rest of the chipyard setup scripts?

jerryz123, 2023-02-17:
This should no longer be an issue with the latest conda-based setup scripts

closed 2023-02-17 by jerryz123