ucb-bar/chipyard #833: Error with thin sbt client: Invalid build URI

tymcauley, 2021-03-22:
**Impact**: other

**Tell us about your environment:**
*Chipyard Version:* dev, hash `262c5c8`
*OS:*
* `Linux 3.10.0-957.27.2.el7.x86_64 #1 SMP Mon Jul 29 17:46:05 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
* `Linux 5.4.0-67-generic #75-Ubuntu SMP Fri Feb 19 18:03:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

*Other:* 

**What is the current behavior?**

When I try to start up the `sbt` thin client (`ENABLE_SBT_THIN_CLIENT=1 make -C sims/vcs start-sbt-server`), I get an error related to the `sbt.workspace` variable from the `sbt` server process.

**What is the expected behavior?**

The `sbt` server should start up.

**Other information**

This seems related to some of the issues I ran into with setting the `sbt.workspace` variable at the bottom of #555. When I try to start up the `sbt` server, I get this:

```
$ ENABLE_SBT_THIN_CLIENT=1 make -C sims/vcs start-sbt-server
...
cd /path/to/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/path/to/chipyard/.java_tmp -jar /path/to/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/path/to/chipyard/tools --client "exit"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] entering *experimental* thin client - BEEP WHIRR
[info] server was not detected. starting an instance
[0m[[0m[0minfo[0m] [0m[0mwelcome to sbt 1.4.4 (Oracle Corporation Java 1.8.0_222)[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chipyardRoot from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project hardfloat from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project rocketConfig from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project firrtl_interpreter from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project treadle from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chisel_testers from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project testchipip from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project icenet from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project hwacha from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project boom from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project cva6 from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project sodor from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project gemmini from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project mdf from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project barstoolsMacros from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project niobium_blocks from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project sim-build from plugins.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading project definition from /path/to/chipyard/sims/firesim/sim/project[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project firesim from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project targetutils from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project midas from build.sbt ...[0m[0J
[0Jjava.lang.RuntimeException: Invalid build URI (no handler available): file:///home/users/tynan/workspace/chisel3/
... (stacktrace follows)
```

So, it's looking in my home directory for `chisel3`, rather than in `/path/to/chipyard/tools`. If I had to guess, that's because while the `sbt` thin client gets `-Dsbt.workspace` set at the command-line, the `sbt` server instance doesn't get that option (or any of the other options). However, if I create a `.sbtopts` file that looks like this:

```
$ cat .sbtopts
-Dsbt.sourcemode=true
-Dsbt.workspace=$PWD/tools
```

... then that `start-sbt-server` command succeeds, and I can run `ENABLE_SBT_THIN_CLIENT=1 make -C sims/vcs verilog`, which interacts with the `sbt` server.

Should we be using a `.sbtopts` file rather than passing those arguments around the `SBT_OPTS` variable in `variables.mk`? Or is there some way to pass arguments to the `sbt` server process? I'm actually also running into a bunch of Java garbage collection performance issues with the `sbt` server, which makes me think it's also not getting the `-Xmx8G -Xss8M -XX:MaxPermSize=256M` settings.

abejgonzalez, 2021-03-28:
Generally, the SBT thin client is really brittle. With that being said, this works on my end.

Running: `ENABLE_SBT_THIN_CLIENT=1 make -C sims/vcs start-sbt-server` on a fresh `dev` (`ed2bfa8`) branch on our development machines.

```
$ ENABLE_SBT_THIN_CLIENT=1 make -C sims/vcs start-sbt-server
make: Entering directory 'chipyard/sims/vcs'
Running with RISCV=chipyard/riscv-tools-install
cd chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=~/.java_tmp -jar chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.ivy.home=~/.ivy2 -Dsbt.global.base=~/.sbt -Dsbt.boot.dire
ctory=~/.sbt/boot/ -Dsbt.sourcemode=true -Dsbt.workspace=chipyard/tools --client "exit"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] entering *experimental* thin client - BEEP WHIRR
[info] server was not detected. starting an instance
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
copying runtime jar...
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[0m[[0m[0minfo[0m] [0m[0mwelcome to sbt 1.4.4 (Ubuntu Java 11.0.10)[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chipyard-build from plugins.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading project definition from chipyard/project[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chipyardRoot from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project hardfloat from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project rocketConfig from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project firrtl_interpreter from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project treadle from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chisel_testers from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project testchipip from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project icenet from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project hwacha from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project boom from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project cva6 from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project sodor from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project gemmini from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project mdf from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project barstoolsMacros from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project sim-build from plugins.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading project definition from chipyard/sims/firesim/sim/project[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project firesim from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project targetutils from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project midas from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chisel3-build from plugins.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading project definition from chipyard/tools/chisel3/project[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project chisel from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project firrtl-build from plugins.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading project definition from chipyard/tools/firrtl/project[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mloading settings for project firrtl from build.sbt ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mresolving key references (55362 settings) ...[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0mset current project to chipyardRoot (in build file:chipyard/)[0m[0J
[0J[0m[[0m[0minfo[0m] [0m[0msbt server started at local://~/.sbt/server/7cdd823c5ea9f1b9c010/sock[0m[0J
[0J[info] terminate the server with `shutdown`
make: Leaving directory 'chipyard/sims/vcs'
```

With the only variables set as:

```
printenv | grep "_OPTS"
JAVA_OPTS=-Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=~/.java_tmp
SBT_OPTS=-Dsbt.ivy.home=~/.ivy2 -Dsbt.global.base=~/.sbt -Dsbt.boot.directory=~/.sbt/boot/
```

Maybe this is a difference in Java versions? I'm running: `openjdk 11.0.10 2021-01-19`

As a side note, you might notice that the thin client doesn't exit properly on a build fail. This is because we aren't using the specific SBT launch script (instead we are launching from the jar file): https://github.com/sbt/sbt/issues/6196. Figured this should be mentioned in text somewhere.

davide-giacomini, 2021-04-06:
I got the same two errors of @tymcauley and @shi27feng in #555. I did what @tymcauley wrote and now I can run `make launch-sbt` and everything goes smooth. I also added the `.sbtopts` file and I can run `sbt` from the root direcotry without problems.

I have an other problem though. If I run in any directory `make launch-sbt` there are no problems. However, if I load the sbt changes from IntelliJ Idea this error shows up:
```
[error] Invalid build URI (no handler available): file:/home/davide/Projects/chipyard/$PWD/tools/chisel3/
```

Moreover, if I try, inside the folder `fpga`, running `make SUB_PROJECT=vcu118 bitstream`, the exact same error typology of @shi27feng shows up. More precisely, the error is the subsequent: 
```
make: *** [/home/davide/Projects/chipyard/common.mk:86: /home/davide/Projects/chipyard/fpga/generated-src/chipyard.fpga.vcu118.VCU118FPGATestHarness.RocketVCU118Config/sim_files.f] Error 1
```
I am simply trying to see how the bitstream is generated for the board above. I am new to Chipyard.

I'm currently on the master branch (latest release), running on an Ubuntu 20.04 machine. 

Thank you very much in advance.

tymcauley, 2021-05-07:
Resolved by #880

PR #880 (2021-05-07): .sbtopts file added + Check for `BINARY/THIN_CLIENT` flags
**Related issue**: #833 

**Type of change**: other enhancement

**Impact**: other

**Release Notes**
It seems like using the `.sbtopts` file helps with the thin SBT client (sets up the Chisel/Firrtl path properly). This PR changes Chipyard to be more similar to Rocket Chip and hopefully fixes #833 (or some subset of its problems). Additionally, this adds small check to make sure the `BINARY` flag is present when doing `run-*` targets.
closed 2021-05-07 by tymcauley