ucb-bar/chipyard #834: A sudden stop when building and installing toolchain

Z-KN, 2021-03-23:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* Chipyard 1.4.0 Release
*OS:* Linux Ubuntu 5.4.0-67-generic #75-Ubuntu SMP Fri Feb 19 18:03:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
*Other:* documentation followed: https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

**What is the current behavior?**
During running the building and installation scripts: './scripts/build-toolchains.sh riscv-tools # for a normal risc-v toolchain', the script exits suddenly leaving these reports:

make[6]: leave directory “/\*\*\*/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib/binutils”
make[5]: leave directory “/\*\*\*/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib/binutils”
make[4]: leave directory “/\*\*\*/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib/binutils”
make[3]: leave directory “/\*\*\*/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib/binutils”
make[2]: leave directory “/\*\*\*/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib”
make[1]: leave directory “/\*\*\*/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib”
mkdir -p stamps/ && touch stamps/build-binutils-newlib

The installation ceases then and there are not any other error messages.
**What is the expected behavior?**
Complete the installation and generate a dicrectory 'riscv-tools-install'.
**Other information**
Previous steps are successfully done.

alonamid, 2021-03-23:
There is not enough information about the error. These are just the final lines of a parallel make run.
There should be additional information further up the log.

Did you make sure all the system dependencies installed successfully?

Z-KN, 2021-03-24:
I fix this problem by re-installing gcc and g++. And now I can run the executable ./simulator-chipyard-RocketConfig.

Your work is complex and amazing. But here I would recommand  adding checks of gcc, g++ and gawk in the script in "1.4. Initial Repository Setup" on the website "https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html". I believe that will make your project better.