ucb-bar/chipyard #844: Documentation for test-chip specific features

jerryz123, 2021-03-31:
**Impact**: docs

**Description**
We are slowly adding testchip-like features into chipyard designs. Some features that would be good to document are

- [ ] Loadmem support in our testchip_tsi implementation
- [ ] TestChip clock/reset handling (Currently the TileResetCtrl registers in testchipip)
- [ ] New testchip_tsi fesvr cmdline args (+init_write, +no_hart0_msip)
- [ ] BootROM and BootAddrReg (from testchipip)
- [ ] CustomBootPin (from testchipip)

colinschmidt, 2021-03-31:
Also explaining what to do when `+init_write` isn't enough.

elamdf, 2024-04-09:
related to loadmem support documentation: https://github.com/ucb-bar/chipyard/issues/1844