ucb-bar/chipyard #847: Cannot create a simulator config with an underscore in its name

JL102, 2021-04-01:
**Impact**: rtl?

**Tell us about your environment:**
*Chipyard Version:*  1.4.0 (2021-01-19)
*OS:*  Ubuntu 20.04 
*Other:* 

I've been creating custom simulator configs inside `generators/chipyard/src/main/scala/config/HeteroConfigs.scala`.
They do not compile properly when the name has an underscore.

For example:
```scala
class BoomD3F8 extends Config(
	new boom.common.WithNSmallBooms(1) ++
	new chipyard.config.AbstractConfig
)
```
The above compiles without issues. However:
```scala
class BoomD3F8_Br8 extends Config(
	new boom.common.WithNSmallBooms(1) ++
	new chipyard.config.AbstractConfig
)
```
The above fails to compile.

**Logs and Stacktrace:**
```
make CONFIG=BoomD3F8_Br8
Running with RISCV=/home/drak/chipyard/esp-tools-install
mkdir -p /home/drak/chipyard/sims/verilator/generated-src/chipyard.TestHarness.BoomD3F8_Br8
cd /home/drak/chipyard && java -Xmx15G -Xss8M -XX:MaxPermSize=256M -jar /home/drak/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/home/drak/chipyard/tools  ";project chipyard; runMain chipyard.Generator  --target-dir /home/drak/chipyard/sims/verilator/generated-src/chipyard.TestHarness.BoomD3F8_Br8 --name chipyard.TestHarness.BoomD3F8_Br8 --top-module chipyard.TestHarness --legacy-configs chipyard:BoomD3F8_Br8"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.4 (Ubuntu Java 11.0.9.1)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/drak/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/drak/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] loading settings for project chisel3-build from plugins.sbt ...
[info] loading project definition from /home/drak/chipyard/tools/chisel3/project
[info] loading settings for project chisel from build.sbt ...
[info] loading settings for project firrtl-build from plugins.sbt ...
[info] loading project definition from /home/drak/chipyard/tools/firrtl/project
[info] loading settings for project firrtl from build.sbt ...
[info] resolving key references (55362 settings) ...
[info] set current project to chipyardRoot (in build file:/home/drak/chipyard/)
[info] set current project to chipyard (in build file:/home/drak/chipyard/)
[info] compiling 1 Scala source to /home/drak/chipyard/tools/firrtl/target/scala-2.12/classes ...
[info] compiling 2 Scala sources to /home/drak/chipyard/generators/rocket-chip/src/target/scala-2.12/classes ...
[warn] /home/drak/chipyard/generators/rocket-chip/src/main/scala/stage/phases/PreElaboration.scala:18:41: trait PreservesAll in package options is deprecated (since FIRRTL 1.4): Use an explicit `override def invalidates` returning false. This will be removed in FIRRTL 1.5.
[warn] class PreElaboration extends Phase with PreservesAll[Phase] with HasRocketChipStageUtils {
[warn]                                         ^
[... 38 lines cut ...]
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error]         at logger.Logger$.$anonfun$makeScope$2(Logger.scala:166)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error]         at logger.Logger$.makeScope(Logger.scala:164)
[error]         at firrtl.options.Stage.transform(Stage.scala:47)
[error]         at firrtl.options.Stage.execute(Stage.scala:58)
[error]         at firrtl.options.StageMain.main(Stage.scala:71)
[error]         at chipyard.Generator.main(Generator.scala)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[error] Caused by: java.lang.ClassNotFoundException: chipyard.Br8
[error]         at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:471)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 31 s, completed Apr 1, 2021, 12:38:15 AM
make: *** [/home/drak/chipyard/common.mk:98: generator_temp] Error 1
```
Presumably, at some point in the code, it splits the config name by underscores and then tries to find chipyard.BoomD3F8 and chipyard.Br8 instead of chipyard.BoomD3F8_Br8. I spent an hour searching for where it does that, but to no avail.

(By the way: The reason I'm giving them weird names is that I'm trying to test the performance of boom cores with different parameters. I got rid of the extra parameters when I copied the code in here.)

alonamid, 2021-04-01:
This is the intended behavior. 

We use underscores to concatenate config fragments into a single config string, and then separate them during elaboration. This is useful when you want to combine a couple of config fragments for quick builds without creating new compound configurations.
You can have a look for example at some example [FireSim config definitions](https://github.com/firesim/firesim/blob/7e32128bea5390ca528052dcdffb13656ee41162/deploy/sample-backup-configs/sample_config_build_recipes.ini#L66).

Yes, it does imply that you cannot use underscores in the names of config fragments. As an aside, the [style guide for Scala classes](https://docs.scala-lang.org/style/naming-conventions.html#classestraits) recommends camel case, which is indeed the style we use for configs.

Regarding that location in the code where the underscore is used as a delimiter, you can [have a look here](https://github.com/ucb-bar/chipyard/blob/master/generators/chipyard/src/main/scala/stage/ChipyardAnnotations.scala)

JL102, 2021-04-01:
Okay, I see. Thanks for the explanation. Any particular reason why underscores have to be used as that delimiter, instead of something like a hyphen?

abejgonzalez, 2021-05-07:
I think we use `_` for legacy reasons... though I can't tell you for sure. You would have to dig back into a ton of old commits to find out the exact reason.

Closing for now since the base question was answered.

closed 2021-05-07 by abejgonzalez