ucb-bar/chipyard #850: The result of testing dhrystone is very unreasonable?

Yujie-Cui, 2021-04-06:
I use chipyard to run dhrystone. When compiling, use make CONFIG=MediumBoomConfig. But the result of the operation is very unreasonable. 
The result is :
Microseconds for one run through Dhrystone: 2085.0
Dhrystones per Second:                       479.6

I did not modify the internal code. So I think the frequency is 1.7Ghz (not sure). Then the score is: 479.6/1757/(1.7*1024). This score is close to 0. Why does this happen? And when I tested Coremark, there was no test result after running for two weeks. Is this reasonable?

Thanks!

jerryz123, 2021-04-15:
I'm not sure what Dhrystone assumes the internal frequency is. You'll have to look in the source for that.

The default coremark iteration count is too high to run in RTL simulation. You have to decrease this to something more reasonable. I think each iteration is about 30 minutes of simulation,

Yujie-Cui, 2021-04-16:
Thank you very much for your reply! I read the code for a long time. I also used vscode to try to search for keywords such as hz, Ghz, khz, fre, but I still couldn't find the place where Boom set the frequency. Can you tell me where the frequency is set or in which file? I use chipyard and compile with make CONFIG=LargeBoomConfig. Thanks!

Yujie-Cui, 2021-04-20:
Does anyone know in which file the frequency is defined? it is important to me. Thank you very much for those who can help.

bobcao3, 2021-04-20:
For dhrystone I remember the frequency is assumed in the benchmark to be 1MHz when it can not find a timer to use, so the dhrystone per second is more like dhrystone per second per megahertz.

Yujie-Cui, 2021-04-21:
Thank you very much for your reply! But doesn't Boom have a fixed frequency like other CPUs? If calculated according to 1Mhz, the DMIPS/MHz of dhrystone is 0.27. But the same configuration I tested on FPGA was 2.6. This seems to be very different? I also tested Coremark in the case of MediumBoomConfig. I also want to solve Coremark's DMIPS/MHz? What is the frequency when testing Coremark?

bobcao3, 2021-04-21:
According to this: https://github.com/riscv/riscv-tests/blob/19bfdab48c2a6da4a2c67d5779757da7b073811d/benchmarks/dhrystone/dhrystone.h#L384

Assuming you are running this benchmark compiled from this source, it assumes a frequency of 1000000 Hz (1MHz). However, this is all assuming you do not have a libc that gives you the `time()` function. It actually does not depend on the frequency you are running at, since DMIPS/MHz basically measures how many dhrystone you can run per 1 million clock cycles. You can try to change the HZ value at that line if you are not sure whether your libc gives you a time function or not. Or you can change that headers and force it to use raw cycle counts

Yujie-Cui, 2021-04-21:
I'm very much obliged to you! This helped me a lot. I just found the setting of -DHZ=100 in the makefile of dhrystone. I am trying to modify this option to see if it depends on this frequency.  Is this also the case with coremark? I am also testing Coremark's DMIPS/MHz. Does Coremark also have its own time measurement?

Yujie-Cui, 2021-04-21:
@bobcao3   You're right. I just re-tested. When I compile dhrystone with **-DHZ=100**, the result is: **Dhrystones per Second: 134.4.** When I use **-DHZ=1000** to compile dhrystone, the result is: **Dhrystones per Second: 1395.9**. The frequency difference is ten times, and the result is about ten times.  Do you know the frequency of Coremark testing?

bobcao3, 2021-04-21:
Not entirely sure. I have not looked at coremark. I think the best thing to
do is to set a frequency value from within chipyard, and if possible get
the results in the unit of cycles.

jerryz123, 2021-05-03:
The riscv-coremark repo in chipyard/software reports cycles.

closed 2021-05-03 by jerryz123