ucb-bar/chipyard #857: scripts/ubuntu-req.sh not configured for Ubuntu 20.04

brendonsb, 2021-04-15:
The Ubuntu requirements script specifies the apt package, python3.6. This package is not available in the current 20.04 repos casuing the script to fail at that point.

hngenc, 2021-04-16:
I think this is an issue with [Chipyard](https://github.com/ucb-bar/chipyard) rather than with Gemmini. `scripts/ubuntu-req.sh` is outside the scope of our project.

hngenc, 2021-04-16:
Oops, sorry, I was on the wrong project page :). Re-opening this

alonamid, 2021-04-16:
Python 3.6 is the minimum requirement, not a "sole" requirement.  Ubuntu 20.04 includes python 3.8, which should satisfy the requirement without need to pull a special package (unlike Ubuntu 16.04 which is the earliest Ubuntu we support).

I agree that the `scripts/ubuntu-req.sh` script is not robust to multiple Ubuntu versions. We will happily accept a PR to make it more robust.

abejgonzalez, 2021-04-28:
There is a current PR trying to see if the change is backwards compatible: #869

abejgonzalez, 2021-05-06:
I think this is solved with #874. Closing for now. Feel free to re-open if there are still issues.

closed 2021-05-06 by abejgonzalez