ucb-bar/chipyard #858: Gemmini (and rocketchip?) can't elaborate

nijssen, 2021-04-16:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* 1.4.0, Hash: 45684f1b875f018a1945476cec9269068e1618ca
*OS:* Linux keylime 5.4.0-45-generic # 49-Ubuntu SMP Wed Aug 26 13:38:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
*Other:* 

**What is the current behavior?**
```
tnijssen@keylime:~/chipyard/sims/verilator$ make CONFIG=gemmini.GemminiHostMiniCore -j8
Running with RISCV=/lscratch/riscv/riscv-mini-1.7
[... snip ...]
[info] running chipyard.Generator --target-dir /soe/tnijssen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.gemmini.Ge
mminiHostMiniCore --name chipyard.TestHarness.gemmini.GemminiHostMiniCore --top-module chipyard.TestHarness --legacy-configs chipy
ard:gemmini.GemminiHostMiniCore
Elaborating design...
[error] java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined
 in Parameters
[error]         ...
[error]         at chipsalliance.rocketchip.config$View.apply(Config.scala:28)
[error]         at chipsalliance.rocketchip.config$View.apply(Config.scala:25)
[error]         at freechips.rocketchip.subsystem.HasConfigurableTLNetworkTopology.$init$(BaseSubsystem.scala:63)
[error]         at freechips.rocketchip.subsystem.BaseSubsystem.<init>(BaseSubsystem.scala:74)
[error]         at chipyard.ChipyardSubsystem.<init>(Subsystem.scala:50)
[error]         at chipyard.ChipyardSystem.<init>(System.scala:24)
[error]         at chipyard.DigitalTop.<init>(DigitalTop.scala:15)
[error]         at chipyard.BuildSystem$$anonfun$$lessinit$greater$1.apply(ChipTop.scala:16)
[error]         at chipyard.BuildSystem$$anonfun$$lessinit$greater$1.apply(ChipTop.scala:16)
[error]         at chipyard.ChipTop.lazySystem$lzycompute(ChipTop.scala:29)
[error]         at chipyard.ChipTop.lazySystem(ChipTop.scala:29)
[error]         at chipyard.ClockingSchemeGenerators$.$anonfun$dividerOnlyClockGenerator$1(Clocks.scala:66)
[error]         at chipyard.ClockingSchemeGenerators$.$anonfun$dividerOnlyClockGenerator$1$adapted(Clocks.scala:62)
[error]         at chipyard.ChipTop.<init>(ChipTop.scala:35)
[error]         at chipyard.BuildTop$$anonfun$$lessinit$greater$1.apply(TestHarness.scala:15)
[error]         at chipyard.BuildTop$$anonfun$$lessinit$greater$1.apply(TestHarness.scala:15)
[error]         at chipyard.TestHarness.<init>(TestHarness.scala:33)
[error]         at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error]         at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error]         at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.
java:45)
[error]         at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
[error]         at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:36)
[error]         ... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 13 s, completed Apr 15, 2021, 11:13:33 PM
make: *** [/soe/tnijssen/chipyard/common.mk:98: generator_temp] Error 1
```

**What is the expected behavior?**
Gemmini is able to generate simulation.

**Other information**
I'm not sure what the best place to file this is. I found [this RocketChip issue](https://github.com/chipsalliance/rocket-chip/issues/2599) which seems related, and applied [that patch](https://github.com/chipsalliance/rocket-chip/pull/2787) to the rocket-chip directory, but no difference.
I'm filing the issue here since RocketChip has (I think) fixed the issue, but the linked commit doesn't exist in the submodule. I just changed the line of code manually.

jerryz123, 2021-04-16:
We do not support the legacy independent configs defined in Gemmini. You should just build GemminiRocketConfig (which is defined in chipyard).

@alonamid @hngenc can one of you delete those configs from gemmini, if they aren't already?

hngenc, 2021-04-16:
@jerryz123 Opened up a PR to remove old configs: https://github.com/ucb-bar/gemmini/pull/108

nijssen, 2021-04-16:
Got it. That seems to work now.

One related question: is it always required to use the Rocket `WithNBigCores` configuration? I tried this configuration and got an exception: (it's just GemminiRocketConfig with the Rocket type changed)
```scala
class GemminiSmallRocketConfig extends Config(
  new gemmini.DefaultGemminiConfig ++
  new freechips.rocketchip.subsystem.With1TinyCore ++
  new chipyard.config.AbstractConfig)
```

```
Elaborating design...
[error] java.lang.IllegalArgumentException: requirement failed: Ports cannot overlap: AddressSet(0x80000000, 0x3fff) AddressSet(0x
80000000, 0xfffffff)                                                                                                              [error]         ...
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.$anonfun$apply$4(AddressDecoder.scala:29)
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.$anonfun$apply$4$adapted(AddressDecoder.scala:28)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.$anonfun$apply$3(AddressDecoder.scala:28)
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.$anonfun$apply$3$adapted(AddressDecoder.scala:28)               [error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.$anonfun$apply$2(AddressDecoder.scala:28)
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.$anonfun$apply$2$adapted(AddressDecoder.scala:27)
[error]         at scala.collection.Iterator.foreach(Iterator.scala:941)
[error]         at scala.collection.Iterator.foreach$(Iterator.scala:941)
[error]         at scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[error]         at freechips.rocketchip.diplomacy.AddressDecoder$.apply(AddressDecoder.scala:27)
[error]         at freechips.rocketchip.tilelink.TLXbar_ACancel$.$anonfun$circuit$98(Xbar.scala:349)                              [error]         at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
[... snip ...]

```

jerryz123, 2021-04-16:
The Medium, Small, and Tiny configs remove important features from rocket, and change the system design in ways that make it incompatible with cache-coherent accelerators

They remove the CPU's FPU, which prevents us from running interesting software without SW emulation of FP (too slow).
TinyConfig replaces the backing memory with a core-local scratchpad, which is not compatible with gemmini.

nijssen, 2021-04-16:
I see, thanks! I'll have to play around some more with it to see if I can shrink down the Rocket size any.

closed 2021-04-16 by nijssen

jerryz123, 2021-04-16:
If you are hitting resource constraints, probably is better to look at reducing Gemmini size.