ucb-bar/chipyard #866: make error: unrecognized command-line option

wky9710, 2021-04-26:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* Chipyard 1.3 Release
*OS:* Linux version 4.15.0-122-generic (buildd@lcy01-amd64-010) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0\-6ubuntu1\~16.04.12)) \#124~16.04.1-Ubuntu SMP
*Other:* documentation followed: [https://docs.boom-core.org/en/latest/](url)

**What is the current behavior?**
the last step, make failed (default config = RocketConfig, so I used the default one, not LargeBoomConfig)
It says "unrecognized command-line option ‘-fbracket-depth=4096’", and "unrecognized command line option ‘-Qunused-arguments’; did you mean ‘-Wunused-argument’?"
for files in *chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/*

According to #691 , I tried to change my compiler from gcc 5.4 to gcc 10.2, but the same question occured.

By the way, *make* was successful, before our server's gcc was accidentally changed by someone. Though we recovered it at once, *make* has never succeeded ever since, so maybe it's because the server's compiler? 

**Other information**
detailed error message: 
```
rm -rf /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig
mkdir -p /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig
verilator --cc --exe -O3 -CFLAGS " -O1 -std=c++11 -I/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR" -CFLAGS "-I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/verilator.h" -LDFLAGS " -L/home/wangkunyu/work/chipyard/sims/verilator -lpthread" /lib/libfesvr.a /home/wangkunyu/work/chipyard/tools/DRAMSim2/libdramsim.a --timescale 1ns/1ps --top-module TestHarness -Wno-fatal --assert --output-split 10000 --output-split-cfuncs 100 --max-num-width 1048576 -f /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/sim_files.common.f /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.top.mems.v /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.harness.mems.v +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\)  -o /home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-RocketConfig -Mdir /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig -CFLAGS "-include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig/VTestHarness.h"
touch /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig/VTestHarness.mk
make VM_PARALLEL_BUILDS=1 -C /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig -f VTestHarness.mk
make[1]: Entering directory '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig'
c++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/verilator.h -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig/VTestHarness.h   -c -o SimDRAM.o /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/SimDRAM.cc
c++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/verilator.h -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig/VTestHarness.h   -c -o SimDTM.o /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/SimDTM.cc
c++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/verilator.h -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig/VTestHarness.h   -c -o SimJTAG.o /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/SimJTAG.cc
c++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O1 -std=c++11 -I/include -I/home/wangkunyu/work/chipyard/tools/DRAMSim2 -D__STDC_FORMAT_MACROS  -DTEST_HARNESS=VTestHarness -DVERILATOR -I/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig.plusArgs -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/verilator.h -include /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig/VTestHarness.h   -c -o SimSerial.o /home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/SimSerial.cc
c++: error：unrecognized command-line option ‘-fbracket-depth=4096’
c++: error：unrecognized command-line option ‘-fbracket-depth=4096’
c++: error：unrecognized command-line option ‘-fbracket-depth=4096’
c++: error：unrecognized command-line option ‘-fbracket-depth=4096’
c++: error：unrecognized command-line option ‘-Qunused-arguments’
VTestHarness.mk:76: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
[... cut ...]
c++: error：unrecognized command-line option ‘-Qunused-arguments’
VTestHarness.mk:78: recipe for target 'SimJTAG.o' failed
make[1]: *** [SimJTAG.o] Error 1
make[1]: Leaving directory '/home/wangkunyu/work/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig/chipyard.TestHarness.RocketConfig'
Makefile:135: recipe for target '/home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-RocketConfig' failed
make: *** [/home/wangkunyu/work/chipyard/sims/verilator/simulator-chipyard-RocketConfig] Error 2
```

Really thanks orz (:3|____)

wky9710, 2021-05-04:
I re-downloaded the chipyard, with version 1.4.0, and met the same error.
If anyone met the same problem, please tell me. 
Really thanks :)

jerryz123, 2021-05-04:
It seems like there is still a gcc versioning issue.

jerryz123, 2023-02-17:
The latest conda-based setup scripts should prevent such issues in the future. closing

closed 2023-02-17 by jerryz123