ucb-bar/chipyard #873: bintray is sunset

timsnyder-siv, 2021-05-03:
**Impact**: other

**Tell us about your environment:**
*Chipyard Version:* 
*OS:* 
*Other:* 

**What is the current behavior?**
Anyone trying to use bintray.com to go get the sbt rpm will hit an error because bintray is no longer.  see https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/

**What is the expected behavior?**
Instructions for installing prereqs should work and not hit errors.

**Other information**

I'm trying some updated instructions for centos in https://github.com/firesim/firesim/pull/764 that I got from https://www.scala-sbt.org/1.x/docs/Installing-sbt-on-Linux.html

alonamid, 2021-05-03:
I believe this is addressed in #870 ?

timsnyder-siv, 2021-05-03:
Yes.  It does look to be fixed in #870 .  Thanks @alonamid

alonamid, 2021-05-06:
Fixed in #874 (since #870 was closed)

closed 2021-05-06 by alonamid