ucb-bar/chipyard #884: How to specify the riscv gcc compiler used when generating simulator-chipyard-LargeBoomConfig

hz1490919302, 2021-05-25:
I found that the default compiler is riscv64-unknown-elf-gcc to generate simulator-chipyard-LargeBoomConfig, but such a simulator cannot run the program compiled by riscv64-unknown-linux-gnu-gcc. The error is: _**warning:tohost and fromhost symbols not in ELF; can't communicate with target**_. I feel that the different compilation tools have caused this problem. Is there any solution?

jerryz123, 2021-05-28:
Those compiled SPEC binaries must be run with pk.

closed 2021-05-28 by jerryz123

hz1490919302, 2021-05-28:
Thank you for your reply. 
I only know that spike pk can run binary files. How can the simulator-chipyard-LargeBoomConfig in the chipyard/sims/verilator directory be used in conjunction with pk?