ucb-bar/chipyard #903: Is it possible to change the dependencies between the buses in the RocketChip subsystem?

SHaBark, 2021-06-18:
Hi! I'm already using the Chipyard framework and successfully generated the Verilog code of the RocketChip core including the subsystem with caches, buses, etc. My goal is to bring everything on an FPGA, so I directly simulate the Verilog code which works well at the moment.

Here is my problem:
In my design I attach some DMA-like device via the FrontEnd bus which has access to the caches inside the RocketChip system. I figured out that during an L2 miss of the core and an access to the external memory via the MemoryBus, the FrontEnd bus is blocked. Blocking of the FrontEnd bus remains until the access via the MemoryBus has finished. That means that there is a dependency between the buses even though they access different addresses. This is a problem for my design because in my case the DMA device also controls the access to the external memory.

Does anybody know where the behavior of the buses is implemented in the CHISEL code? Or has somebody an idea of how to (at least partly) disable the mentioned dependency?

jerryz123, 2021-06-18:
Is there some access from the FrontBus that is also trying to access L2, which is causing the dependency?

SHaBark, 2021-06-18:
Right. The access from the FrontBus tries to access L2 and could cause the dependency.
But the dependency also occurs when the core accesses L2. In case of an L2 miss another access follows through the MemoryBus. During this time the FrontBus is blocked. In my design this is a problem because the DMA device manages the access from MemoryBus to external memory.

jerryz123, 2021-06-25:
That setup would inherently create a circular dependency, causing deadlock issues.
What devices does your thing need to access on the system? Perhaps attaching it to the Frontbus is unnecessary, and you can connect it more closely to whatever its trying to access.

SHaBark, 2021-06-25:
Thanks for your comment! I indeed got the deadlock. This is why I'm asking how to change the behavior of the buses.
My device at the FrontBus is a DMA controller which wants to access the internal caches of the Rocket core. I don't necessarily need to go via the FrontBus, but this was the easiest solution so far. I would prefer to directly get access to the caches but that would be a massive change of the Rocket subsystem.

alonamid, 2021-06-28:
Why not connect to the system bus or the periphery bus?

alonamid, 2021-06-28:
Either way, it seems that there is a higher likelihood that you block is not fully complying to the TileLink protocol assumption when connecting to the bus