ucb-bar/chipyard #904: RiscV-Sodor core tracer.py information in other simulated cores??

Kian75, 2021-06-23:
Hi... I have recently started to work with chipyard framework, so my question can be a basic one...
In riscv-sodor repository there is a python code in [https://github.com/ucb-bar/riscv-sodor/tree/master/scripts](url) which is used in simulation of Sodor processor and adds the information about CPI, percentage of each instruction in simulated benchmark and.... 
My question is how can I use this feature in simulation of other cores? for example I go through the simulation flow of Rocket Core and all of the simulations have passed. Is there a way to see the mentioned statistics in .out files?

jerryz123, 2021-06-23:
We use those scripts in cs 152 only. For more advanced cores like Rocket or boom generally we prefer to directly instrument the code with rdcycle or rdinstret instructions

Kian75, 2021-06-23:
Okay...  Thank you for your quick answer...
Actually I don't know that part of rdcycle and rdinstret that you mentioned :((....
but still if I simulate the Sodor Core in Chipyard with benchmarks available, I can see those statistics at the end of .output file?
I am asking because I tried to go through simulation flow of Sodor but I get an error 137 about firrtle-temp which I think it's about limited Ram of my laptop...

jerryz123, 2023-02-17:
Closing due to inactivity.

closed 2023-02-17 by jerryz123