ucb-bar/chipyard #907: Tracing not working for SmallBoomConfig

galsanka, 2021-06-25:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* 1.5.0, Hash: b5d013
*OS:* `centos-release-7-9.2009.1.el7.centos.x86_64` 
*Other:* 

**What is the current behavior?**
Baremetal riscv programs (i.e. https://chipyard.readthedocs.io/en/latest/Software/Baremetal.html) run correctly on verilator and vsim-based simulators, but instruction-by-instruction logs are not being printed.

**What is the expected behavior?**
Executables run correctly and output traces.

**Other information**

- Ran `make debug CONFIG=SmallBoomConfig` in both `sims/vsim` and `sims/verilator` directories. Resulting executable runs `.riscv` files correctly, but doesn't output an instruction-by-instruction log.
- Also used the `+verbose` flag, as instructed by this link: https://chipyard.readthedocs.io/en/latest/Advanced-Concepts/Debugging-RTL.html.
- Tried: adding a new configuration in `generators/chipyard/src/main/scala/config/BoomConfigs.scala`, which unfortunately didn't work either. The class ended up looking like ```
class SmallBoomConfigTrace extends Config(
  new chipyard.config.WithTraceIO ++
  new boom.common.WithNSmallBooms(1) ++                          // small boom config
  new chipyard.config.AbstractConfig)
```

jerryz123, 2021-06-30:
You must add the WithBoomCommitLogPrintf fragment to your config

galsanka, 2021-06-30:
That fixed it. Thank you so much!

Quick question: what version of vcs is officially supported?

closed 2021-07-01 by galsanka

nishanthsmurthy24, 2023-07-19:
I have this configuration 

```
class SmallBoomConfig extends Config(
  new boom.common.WithNSmallBooms(1) ++
  new boom.common.WithBoomCommitLogPrintf ++                          // small boom config
  new chipyard.config.AbstractConfig)
```

and when I run

`make CONFIG=SmallBoomConfig run-binary BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-simple`

using verilator, I obtain this in my output file.

```
using random seed 1689833396
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 33026
== Loading device model file '/root/chipyard/generators/testchipip/src/main/resources/dramsim2_ini/DDR3_micron_64M_8B_x4_sg15.ini' == 
== Loading system model file '/root/chipyard/generators/testchipip/src/main/resources/dramsim2_ini/system.ini' == 
===== MemorySystem 0 =====
CH. 0 TOTAL_STORAGE : 4096MB | 1 Ranks | 16 Devices per rank
DRAMSim2 Clock Frequency =666666666Hz, CPU Clock Frequency=100000000Hz
*** PASSED *** Completed after 4444 cycles
```
Am I missing something that is not giving me detailed trace output? Any help is much appreciated.

jerryz123, 2023-07-19:
`WithBoomCommitLogPrintf` must be ordered before `WithNSmallBooms`

nishanthsmurthy24, 2023-07-19:
> `WithBoomCommitLogPrintf` must be ordered before `WithNSmallBooms`

That worked. But the Trace ouput does not contain cycle specific instruction.

My Objective is to check at which cycle `ecall` instruction was executed on different cores in Chipyard. This is easy to obtain for `Rocket` configurations.

Is there a work around for `BOOM` configurations?

jerryz123, 2023-07-19:
You have to add the cycle information to the printout in the boom implementation.

nishanthsmurthy24, 2023-07-19:
I did some digging and found this file,

`generators/boom/src/main/scala/lsu/tracegen.scala`

is this the place where the trace is getting generated? Correct me if I am wrong. 

Can you point out the location where this print statement of cycles should be implemented?

jerryz123, 2023-07-19:
No, its the print statements in core.scala. look for `COMMIT_LOG_PRINTF`.

YiKangOY, 2024-03-14:
@nishanthsmurthy24 Hi, I also want to print the instruction being executed every cycle. Have you successfully implemented the printing? If so, may I ask how did you change your core.scala?