ucb-bar/chipyard #910: [error] Invalid build URI (no handler available): file:/home/hshan/chipyard/sims/firesim/sim/

shz0116, 2021-06-28:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* 
*OS:* 
*Other:* 

I am trying to follow the ISCA21 tutorial.
https://fires.im/isca21-slides-pdf/03_building_custom_socs.pdf
On slides 12, when I ran "make CONFIG=HeteroConfig" under /home/hshan/chipyard/sims/verilator,
I got the following error:
hshan@sjchpc01-ux:/home/hshan/chipyard/sims/verilator$ make CONFIG=HeteroConfig 
Running with RISCV=/home/hshan/chipyard/esp-tools-install
mkdir -p /home/hshan/chipyard/sims/verilator/generated-src/chipyard.TestHarness.MytestConfig
cd /home/hshan/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/home/hshan/chipyard/.java_tmp -jar /home/hshan/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/home/hshan/chipyard/tools  ";project chipyard; runMain chipyard.Generator  --target-dir /home/hshan/chipyard/sims/verilator/generated-src/chipyard.TestHarness.MytestConfig --name chipyard.TestHarness.MytestConfig --top-module chipyard.TestHarness --legacy-configs chipyard:MytestConfig"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.9 (Ubuntu Java 11.0.11)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/hshan/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
java.lang.RuntimeException: Invalid build URI (no handler available): file:/home/hshan/chipyard/sims/firesim/sim/
        at scala.sys.package$.error(package.scala:30)
        at sbt.internal.Load$.$anonfun$builtinLoader$1(Load.scala:483)
        at sbt.internal.BuildLoader.$anonfun$apply$3(BuildLoader.scala:244)
        at scala.Option.getOrElse(Option.scala:189)
        at sbt.internal.BuildLoader.apply(BuildLoader.scala:244)
        at sbt.internal.Load$.loadURI$1(Load.scala:548)
        at sbt.internal.Load$.loadAll(Load.scala:564)
        at sbt.internal.Load$.loadURI(Load.scala:494)
        at sbt.internal.Load$.load(Load.scala:473)
        at sbt.internal.Load$.$anonfun$apply$1(Load.scala:242)
        at sbt.internal.Load$.timed(Load.scala:1382)
        at sbt.internal.Load$.apply(Load.scala:242)
        at sbt.internal.Load$.defaultLoad(Load.scala:55)
        at sbt.BuiltinCommands$.liftedTree1$1(Main.scala:934)
        at sbt.BuiltinCommands$.doLoadProject(Main.scala:934)
        at sbt.BuiltinCommands$.$anonfun$loadProjectImpl$2(Main.scala:887)
        at sbt.Command$.$anonfun$applyEffect$4(Command.scala:150)
        at sbt.Command$.$anonfun$applyEffect$2(Command.scala:145)
        at sbt.Command$.process(Command.scala:189)
        at sbt.MainLoop$.$anonfun$processCommand$5(MainLoop.scala:243)
        at scala.Option.getOrElse(Option.scala:189)
        at sbt.MainLoop$.process$1(MainLoop.scala:243)
        at sbt.MainLoop$.processCommand(MainLoop.scala:276)
        at sbt.MainLoop$.$anonfun$next$5(MainLoop.scala:163)
        at sbt.State$StateOpsImpl$.runCmd$1(State.scala:289)
        at sbt.State$StateOpsImpl$.process$extension(State.scala:325)
        at sbt.MainLoop$.$anonfun$next$4(MainLoop.scala:163)
        at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
        at sbt.MainLoop$.next(MainLoop.scala:163)
        at sbt.MainLoop$.run(MainLoop.scala:144)
        at sbt.io.Using.apply(Using.scala:27)
        at sbt.MainLoop$.runWithNewLog(MainLoop.scala:112)
        at sbt.MainLoop$.runAndClearLast(MainLoop.scala:66)
        at sbt.MainLoop$.runLoggedLoop(MainLoop.scala:51)
        at sbt.MainLoop$.runLogged(MainLoop.scala:42)
        at sbt.StandardMain$.runManaged(Main.scala:193)
        at sbt.xMain$.$anonfun$run$8(Main.scala:102)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
        at scala.Console$.withIn(Console.scala:230)
        at sbt.internal.util.Terminal$.withIn(Terminal.scala:560)
        at sbt.internal.util.Terminal$.$anonfun$withStreams$1(Terminal.scala:350)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
        at scala.Console$.withOut(Console.scala:167)
        at sbt.internal.util.Terminal$.$anonfun$withOut$2(Terminal.scala:550)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
        at scala.Console$.withErr(Console.scala:196)
        at sbt.internal.util.Terminal$.withOut(Terminal.scala:550)
        at sbt.internal.util.Terminal$.withStreams(Terminal.scala:350)
        at sbt.xMain$.run(Main.scala:86)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at sbt.internal.XMainConfiguration.run(XMainConfiguration.scala:83)
        at sbt.xMain.run(Main.scala:46)
        at xsbt.boot.Launch$$anonfun$run$1.apply(Launch.scala:109)
        at xsbt.boot.Launch$.withContextLoader(Launch.scala:128)
        at xsbt.boot.Launch$.run(Launch.scala:109)
        at xsbt.boot.Launch$$anonfun$apply$1.apply(Launch.scala:35)
        at xsbt.boot.Launch$.launch(Launch.scala:117)
        at xsbt.boot.Launch$.apply(Launch.scala:18)
        at xsbt.boot.Boot$.runImpl(Boot.scala:41)
        at xsbt.boot.Boot$.main(Boot.scala:17)
        at xsbt.boot.Boot.main(Boot.scala)
[error] Invalid build URI (no handler available): file:/home/hshan/chipyard/sims/firesim/sim/
[error] Use 'last' for the full log.
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
make: *** [/home/hshan/chipyard/common.mk:106: generator_temp] Error 1

Any idea ? I checked other similar issues, but seems different from my problem.
Thanks

shz0116, 2021-06-28:
Checked issue#555, but solution there seems not working.
Tried different verilator version 4.034, 4,204 showing same errors.
My git is version 2.25.1, should not be a problem neither.

alonamid, 2021-06-28:
I have just tried cloning from scratch, adding the `HeteroConfig` from the tutorial, and building it in `sims/verilator`.
With the exception that `freechips.rocketchip.subsystem.WithNLargeCores(1)` in slide 10 should be `freechips.rocketchip.subsystem.WithNBigCores(1)`, everything else seems to have built fine. 
Hence, I'm suspecting this might be an environment issue. I ran it on both Ubuntu 18.04 and CentOS 7, and it worked.
Do you have all the system requirements listed here? https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html

davidbiancolin, 2021-06-28:
Can you also confirm there's a build.sbt here:
> /home/hshan/chipyard/sims/firesim/sim

shz0116, 2021-06-28:
No files under /home/hshan/chipyard/sims/firesim/sim.
I need to create one manually ?

davidbiancolin, 2021-06-28:
The FireSim git submodule wasn't initialized properly. You'll have you check the log from init-submodules and see what happened.

shz0116, 2021-06-28:
Thanks. Let me check where went wrong.

shz0116, 2021-06-28:
I found the problem. As @alonamid suggested, I did not install all the packages.

closed 2021-06-28 by alonamid

shz0116, 2021-06-28:
However, a following problem is it complains the package is missing.

hshan@sjchpc01-ux:/home/hshan/chipyard/sims/verilator$ make CONFIG=MytestConfigs
Running with RISCV=/home/hshan/chipyard/esp-tools-install
mkdir -p /home/hshan/chipyard/sims/verilator/generated-src/chipyard.TestHarness.MytestConfigs
cd /home/hshan/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/home/hshan/chipyard/.java_tmp -jar /home/hshan/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/home/hshan/chipyard/tools  ";project chipyard; runMain chipyard.Generator  --target-dir /home/hshan/chipyard/sims/verilator/generated-src/chipyard.TestHarness.MytestConfigs --name chipyard.TestHarness.MytestConfigs --top-module chipyard.TestHarness --legacy-configs chipyard:MytestConfigs"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.9 (Ubuntu Java 11.0.11)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/hshan/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/hshan/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] loading settings for project chisel3-build from plugins.sbt ...
[info] loading project definition from /home/hshan/chipyard/tools/chisel3/project
[info] loading settings for project chisel from build.sbt ...
[info] loading settings for project firrtl-build from plugins.sbt ...
[info] loading project definition from /home/hshan/chipyard/tools/firrtl/project
[info] loading settings for project firrtl from build.sbt ...
[info] resolving key references (55373 settings) ...
[info] set current project to chipyardRoot (in build file:/home/hshan/chipyard/)
[info] set current project to chipyard (in build file:/home/hshan/chipyard/)
[info] running chipyard.Generator --target-dir /home/hshan/chipyard/sims/verilator/generated-src/chipyard.TestHarness.MytestConfigs --name chipyard.TestHarness.MytestConfigs --top-module chipyard.TestHarness --legacy-configs chipyard:MytestConfigs
[error] (run-main-0) java.lang.Exception: Unable to find part "chipyard.MytestConfigs" from "ArraySeq(chipyard.MytestConfigs)", did you misspell it or specify the wrong package path?
[error] java.lang.Exception: Unable to find part "chipyard.MytestConfigs" from "ArraySeq(chipyard.MytestConfigs)", did you misspell it or specify the wrong package path?
[error]         at Chisel.package$throwException$.apply(compatibility.scala:439)
[error]         at freechips.rocketchip.util.HasRocketChipStageUtils.$anonfun$getConfig$1(GeneratorUtils.scala:19)
[error]         at scala.collection.IndexedSeqOptimized.foldRight(IndexedSeqOptimized.scala:65)
[error]         at scala.collection.IndexedSeqOptimized.foldRight$(IndexedSeqOptimized.scala:72)
[error]         at scala.collection.mutable.ArraySeq.foldRight(ArraySeq.scala:47)
[error]         at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig(GeneratorUtils.scala:14)
[error]         at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig$(GeneratorUtils.scala:13)
[error]         at freechips.rocketchip.stage.phases.PreElaboration.getConfig(PreElaboration.scala:18)
[error]         at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:29)
[error]         at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:18)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:248)
[error]         at firrtl.options.Translator.transform$(Phase.scala:248)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:278)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[... cut ...]
 26   // Uncomment out one (or multiple) of the lines below, and choose
 27   // how many cores you want.
 28   new freechips.rocketchip.subsystem.WithNBigCores(1) ++    // Specify we want some number of Rocket cores
 29   new boom.common.WithNSmallBooms(1) ++                     // Specify we want some number of BOOM cores
 30
 31   // CUSTOMIZE the L2
 32   // Uncomment this line, and specify a size if you want to have a L2
 33   new freechips.rocketchip.subsystem.WithInclusiveCache(capacityKB=256) ++
 34
 35   new chipyard.config.AbstractConfig
 36 )

YiKangOY, 2023-06-14:
> However, a following problem is it complains the package is missing.
>
> hshan@sjchpc01-ux:/home/hshan/chipyard/sims/verilator$ make CONFIG=MytestConfigs Running with [...]

Hi, I have met the same problem as yours. May I ask how do you solve it?