ucb-bar/chipyard #911: Add xiangshan core

HKalbasi, 2021-06-28:
There is a new risc v core named Xiangshan ([github](https://github.com/OpenXiangShan/XiangShan) ) written in chisel. Is it possible to have it in chipyard? Will it be useful for chipyard and/or xiangshan and worth the needed efforts?

abejgonzalez, 2021-08-23:
I don't think Chipyard folks are opposed to having a new core integrated w/ Chipyard. I think the main issue is maintenance and upkeep. We would need someone to own the code (i.e. we need to be able to reach out to them if something needs to be bumped/fixed/upgraded), add CI, etc. With that being said, I think comparing cores would be great!

HKalbasi, 2021-09-05:
I opened [an issue in xiangshan](https://github.com/OpenXiangShan/XiangShan/issues/855) at the same time. From their comment I think they have works with more priority than maintaining it in Chipyard, and a "cleanup" is needed before integration, but in long-term I see Chipyard can make XiangShan more discover-able and removing some headache in using the core, and in other direction, Chipyard users will have another option, and more options is better.

So if you can provide some instructions on that thread about what should be done to make this integration, Xiangshan team (or community) may consider it in future.

jerryz123, 2025-03-30:
I believe XS has diverged from the rocket-chip/chisel we use, so integrating it as a chisel project will not be easy.

redpanda3, 2025-03-31:
Yeah...I've discussed this with Jerry Zhao, maybe about 3 years ago? 

It's nice to see more different kinds of CPUs in the same integration environment so I created this integration solution. 

However, integrating it as a chisel project was not that easy(too much dependency conflicts, maybe AI can reduce some of the work?), so I gave up this kind of trial. Actually integration of xiangshan is not the same as nvdla, xiangshan is a master node in tilelink. So I reviewed with Xiangshan developers for some time.

Best,
Redpanda3

> Thank you for the initiative! Regarding "so if you can provide some instructions on that thread about what should be done to make this integration" I believe the steps might be:
> 
> * Integrate the core into a Chipyard repo if not already
>   
>   * (including your core repository as a generator submodule, including the corresponding sbt file changes, create some example Configs in order to be able to test the core, for example the RISC-V ISA tests since this is a RISC-V core, etc.. feel free to ask questions!)
> * Submit a Chipyard PR with the the core integrated - or a draft PR with what steps remain or you'd like advice on.
> * Write some documentation for the core
>   
>   * (or adapt your existing documentation to make it available from the main Chipyard docs)
> * Show the core is functional and document limitations
>   
>   * (such as whether it successfully passes simulation, whether it's silicon proven, whether it's been PaR'd / physical design-ed..)
> * I see the project is still active, but it would still be nice to have a point of contact as mentioned above by Abe. If that's impossible, there might be a route to having the core be part of a branch first?
> 
> Regarding [@redpanda3](https://github.com/redpanda3)'s integration - I see you're basing it off of `https://github.com/ucb-bar/nvdla-wrapper`. Note that that wrapper and add-ons (such as the pre-processing done in the [Makefile](https://github.com/ucb-bar/nvdla-wrapper/blob/master/src/main/resources/Makefile) ) are for Verilog IP, and you don't need to blackbox Chisel IP if the core is in Chisel.
> 
> We have a taped-out integration based off that wrapper in one of our experimental classes (look for "OFOT generator"), which looks like:
> 
> * https://github.com/ucb-eecs151tapeout/ofot-generator
> * https://github.com/ucb-eecs151tapeout/ofot-chipyard
> 
> [Website - https://151tapeout.berkie.ee/](https://151tapeout.berkie.ee/) ["Labs" which contain some documentation - https://github.com/ucb-eecs151tapeout/eecs151t-labs](https://github.com/ucb-eecs151tapeout/eecs151t-labs)
> 
> However again, in this case we are again using Verilog IP - once you return to this project, you might want to check which parts of the code are Verilog-specific and which aren't.
> 
> Belated but hope this helps!

hkalbasi, 2026-09-01:
Closing due to lack of interest.

closed 2026-09-01 by hkalbasi