ucb-bar/chipyard #912: WithJustOneBus Parameter not working

RaphaelKlink, 2021-06-30:
Hello,
i try to create the smallest possible Chipyard Configuration.
During this task i stumbled upon the WithJustOneBus Parameter. This reduces the Busses to just one the System Bus.
I tried different configurations. But i could not get it to work, as some modules always needs the Control Bus.

The last one i tried is:

class OneBusConfig extends Config(
  new chipyard.iobinders.WithTieOffInterrupts ++                       // tie off top-level interrupts
  new chipyard.iobinders.WithSimDebug ++                               // tie off debug port
  new freechips.rocketchip.subsystem.WithNoMemPort ++
  new freechips.rocketchip.subsystem.WithNMemoryChannels(0) ++
  new freechips.rocketchip.subsystem.WithNoSlavePort ++
  new freechips.rocketchip.subsystem.WithNoMMIOPort ++
  new freechips.rocketchip.subsystem.WithNBanks(0) ++                   
  new freechips.rocketchip.subsystem.WithIncoherentTiles ++
  new freechips.rocketchip.subsystem.WithNExtTopInterrupts(0) ++ // no external interrupts
  new freechips.rocketchip.subsystem.WithScratchpadsOnly ++            // Use ScratchPad instead of RAM
  new freechips.rocketchip.subsystem.With1TinyCore ++               // single rocket-core
  new freechips.rocketchip.subsystem.WithJustOneBus ++
  new freechips.rocketchip.system.BaseConfig
)

Could someone of you help?
Which modules do i need to remove also?

Greetings Raphael Klink

jerryz123, 2021-06-30:
What version of Chipyard are you using?

A design with hierarchical buses should require fewer resources than one which connects all the devices to a shared bus.

sammhho, 2021-08-17:
> Hello,
> i try to create the smallest possible Chipyard Configuration.
> During this task i stumbled upon the WithJustOneBus Parameter. This reduces the Busses to just one the System Bus.
> I tried [...]

Hi @RaphaelKlink,

I'm not from the team, but I think I got part of your answer.

I can confirm that the `WithJustOneBus` is working as expected, by having a `BaseSubSystem` and adding only modified `CanHavePeripheryGCD` and `CanHavePeripheryInitZero` to the `sbus`.

You'd really want to try not basing upon `BaseConfig`, as it's based on `BaseSubsystemConfig`, which has quite a few things using the `cbus`.
Alternatively, you might at least want to replace the default `PLICParams`, `CLINTParams` and `DefaultDebugModuleParams` as these seems to use the `cbus` and is replaced with `None` as demonstrated in `GroundTestBaseConfig` in `generators/rocket-chip/src/main/scala/groundtest/Configs.scala`.
Replacing to `None` may not make sense in your case since you actually do want a rocket core, so you might need to modify those params to use sbus whenever it use cbus.

I'm not sure these make sense, can someone from the team confirm?

jerryz123, 2021-08-17:
Your understanding seems to be correct. A lot of the default config fragments assume the hierarchical bus topology. If you want to build a functionally equivalent system with a single-bus topology. you must write fragments to place devices off the sbus.

RaphaelKlink, 2021-08-17:
Thank you for help.

Yes i was using an older version of Chipyard.

Thank you for the clarification.

closed 2021-08-17 by RaphaelKlink