ucb-bar/chipyard #914: VCS simulation not working

galsanka, 2021-07-02:
**Impact**: software

**Tell us about your environment:**
*Chipyard Version:* 1.5.0, Hash: b5d013
*OS:* `centos-release-7-9.2009.1.el7.centos.x86_64` 
*Other:* VCS version Q-2020.03-SP1-1

**What is the current behavior?**
Vsim-based BOOM simulators display the following error when running the built `./simv-chipyard-*` executable:
```
TARGET (RISC-V BINARY) OPTIONS
  These are the options passed to the program executing on the emulated RISC-V
  microprocessor.

terminate called after throwing an instance of 'std::invalid_argument'
  what():  No binary specified (Did you forget it? Did you forget '+permissive-off' if r

Error-[DPI-UED] C++ Exception detected
  Import DPI routine invoked at file
  '/home/grads/g/georges/fuzz/chipyard/sims/vcs/generated-src/chipyard.TestHarness.Smallline
  53) has C++ exceptions not caught. C++ exceptions shall not propagate out of
  any imported subroutine.
  Fix it in DPI-C code before running simulation.

```
When running the executable with an `-h` flag, the following error is displayed at the beginning of the output:
```
Warning-[RT_UO] Unsupported option
  Unsupported option '-h' is ignored
```

**What is the expected behavior?**
Simulation executables based on vcs run correctly with no errors thrown.

**Other information**

- The equivalent verilator-based executable works fine

jerryz123, 2021-07-03:
Are you manually invoking the VCS binary? Or using the built in make commands. 

Either way this is a bug, since VCS should behave same as Verilator. But using the built in Make commands may sidestep this issue.

galsanka, 2021-07-03:
I have been using the built in make commands, with flags/arguments that worked for the verilator equivalent.

jerryz123, 2021-07-03:
Can you post the full command ?

galsanka, 2021-07-03:
`make CONFIG=SmallBoomConfig -j $(nproc)`

jerryz123, 2021-07-03:
That is the command to build the simulator. What command are you using to run the simulation.

galsanka, 2021-07-03:
`simv-chipyard-SmallBoomConfig`

jerryz123, 2021-07-03:
What is hello.v? That doesn't seem like a riscv binary

galsanka, 2021-07-03:
Sorry. That was a typo on my part. Any riscv binary compiled with https://chipyard.readthedocs.io/en/latest/Software/Baremetal.html gives the same result when combined with +verbose. The simulator works when only given a `.riscv` file, but doesn't when using any flag, including `-h` or `+verbose`

alonamid, 2021-07-07:
I think @jerryz123 was talking about [execution make commands](https://chipyard.readthedocs.io/en/latest/Simulation/Software-RTL-Simulation.html#fast-memory-loading) such as `make run-binary` and `make run-binary-hex`.
If you are invoking the simulator directly and trying to add VCS flags, you might want to add the `+permissive` and `+permissive-off` flags before an after you VCS plug args.

WegdanAli13, 2022-07-12:
Hello, I am facing the same error with vcs when trying the vlsi flow as I want to simulate my design. I use the command "make sim-rtl CONFIG=TinyRocketConfig BINARY=$RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv32ui-p-simple" as in the ASAP7 example . Can you help me if you figured out anything ?