ucb-bar/chipyard #917: dromajo bootrom issue

kabylkas, 2021-07-11:
Hi!

It seems bootrom directory has been removed from the latest chipyard repo. However, dromajo is still trying to get the bootrom from that path.
https://github.com/ucb-bar/chipyard/blob/b5d013190d637e634113cb5179f8c8885df1945a/tools/dromajo/dromajo.mk#L10

Should this be changed to a `DROMAJO_ROM = $(build_dir)/bootrom.rv64.img`?

Thanks!

kabylkas, 2021-07-11:
For the purposes of the forward progress, I speculated that `DROMAJO_ROM` should point to `$(build_dir)/bootrom.rv64.img` 😊 When I do this, the cosimulation fails when executing bootrom code (see below). Before digging into the issue, could you please confirm I am doing everything right 😀 Thanks!

```
testing $random 18faeb31 seed 3990867110
== Loading device model file '/chipyard/generators/testchipip/src/main/resources/dramsim2_ini/DDR3_micron_64M_8B_x4_sg15.ini' ==
== Loading system model file '/chipyard/generators/testchipip/src/main/resources/dramsim2_ini/system.ini' ==
===== MemorySystem 0 =====
CH. 0 TOTAL_STORAGE : 4096MB | 1 Ranks | 16 Devices per rank
DRAMSim2 Clock Frequency =666666666Hz, CPU Clock Frequency=100000000Hz
CORE0: 3 0x0000000000010040 (0x00000517) x10 0x0000000000010040 auipc   a0, 0x0
CORE0: 3 0x0000000000010044 (0xfc050513) x10 0x0000000000010000 addi    a0, a0, -64
csr_read: hartid=0 csr=0x305 val=0x0
csr_write: hardid=0 csr=0x305 val=0x0000000000010000
CORE0: 3 0x0000000000010048 (0x30551073)                        csrw    mtvec, a0
csr_read: hartid=0 csr=0x301 val=0x94112d
CORE0: 3 0x000000000001004c (0x301022f3) x5  0x800000000094112d csrr    t0, misa
CORE0: 3 0x0000000000010050 (0x4122d293) x5  0xffffe00000000025 srai    t0, t0, 18
CORE0: 3 0x0000000000010054 (0x0012f293) x5  0x0000000000000001 andi    t0, t0, 1
CORE0: 3 0x0000000000010058 (0x00028463)                        beqz    t0, pc + 8
csr_read: hartid=0 csr=0x303 val=0x0
csr_write: hardid=0 csr=0x303 val=0x0000000000000000
CORE0: 3 0x000000000001005c (0x30301073)                        csrw    mideleg, zero
CORE0: 3 0x0000000000010060 (0x00800513) x10 0x0000000000000008 li      a0, 8
csr_read: hartid=0 csr=0x304 val=0x0
csr_write: hardid=0 csr=0x304 val=0x0000000000000008
CORE0: 3 0x0000000000010064 (0x30451073)                        csrw    mie, a0
csr_read: hartid=0 csr=0x300 val=0x1800
csr_write: hardid=0 csr=0x300 val=0x0000000a00001808
CORE0: 3 0x0000000000010068 (0x30052073)                        csrs    mstatus, a0
CORE0: 3 0x000000000001006c (0x10500073)                        wfi
[DEBUG] DUT raised interrupt 3
[DEBUG] Interrupt: MIP <- 3: Now MIP = 8
get_irq_mask: mip=0x8 mie=0x8 mideleg=0x0
raise_interrupt: irq=3 priv=3 pc=10070 hartid=0
hartid=0: exception interrupt #3, epc 0x0000000000010070
CORE0: 3 0x0000000000010000 (0x020005b7) x11 0x0000000002000000 lui     a1, 0x2000
csr_read: hartid=0 csr=0xf14 val=0x0
CORE0: 3 0x0000000000010004 (0xf1402573) x10 0x0000000000000000 csrr    a0, mhartid
CORE0: 3 0x0000000000010008 (0x00050463)                        beqz    a0, pc + 8
CORE0: 3 0x0000000000010010 (0x00458613) x12 0x0000000002000004 addi    a2, a1, 4
CORE0: 3 0x0000000000010014 (0x00100693) x13 0x0000000000000001 li      a3, 1
clint_write: MSIP access for hartid:1 which is beyond ncpus
CORE0: 3 0x0000000000010018 (0x00d62023)                        sw      a3, 0(a2)
CORE0: 3 0x000000000001001c (0x00460613) x12 0x0000000002000008 addi    a2, a2, 4
clint_read: MSIP access for hartid:1 which is beyond ncpus
CORE0: 3 0x0000000000010020 (0xffc62683) x13 0x0000000000000000 lw      a3, -4(a2)
CORE0: 3 0x0000000000010024 (0xfe069ae3)                        bnez    a3, pc - 12
CORE0: 3 0x0000000000010028 (0x05c0006f)                        j       pc + 0x5c
CORE0: 3 0x0000000000010084 (0x0005a023)                        sw      zero, 0(a1)
CORE0: 3 0x0000000000010088 (0x00004537) x10 0x0000000000004000 lui     a0, 0x4
riscv_cpu_read_memory: invalid physical address 0x0000000000004000
hartid=0 priv: 3 exception fault_load, epc 0x000000000001008c
hartid=00           tval 0x0000000000004000
CORE0: 3 0x0000000000010000 (0x020005b7) x11 0x0000000002000000 lui     a1, 0x2000
[error] EMU PC 0000000000010000, DUT PC 000000000001008c
[error] EMU INSN 020005b7, DUT INSN 00056503
[error] EMU WDATA 0000000002000000, DUT WDATA 0000000080000000
[error] EMU MSTATUS a00001800, DUT MSTATUS 00000000
[error] DUT pending exception -1 pending interrupt -1 
```

The above trace is the output of the following command:
```
make CONFIG=DromajoBoomConfig ENABLE_DROMAJO=1 BINARY=$(path_to_bins)/rv64ui-p-and run-binary
```

jerryz123, 2021-07-11:
We added a register mapped device at 0x4000 that is accessed at boot.
Dromajo needs to be modified to add a device here.

kabylkas, 2021-07-16:
Adding the device in dromajo fixed this, thank you! 

However, right now, I am hardcoding the address and the return value of the device to mimic BOOM. I was thinking how to make this cleaner. 

I noticed that dromajo is referencing dromajo_params.h for some values (see below). I was wondering if this file and the parameters are auto generated? Or it is just a hardcoded values? If the latter is true, maybe I could add more defines to facilitate the above mentioned device?

```c
#ifndef DROMAJO_PARAMS_H
#define DROMAJO_PARAMS_H

#define DROMAJO_RESET_VECTOR "0x10040"
#define DROMAJO_MMIO_START "0x20000"
#define DROMAJO_MMIO_END "0x80000000"
#define DROMAJO_MEM_SIZE "0x100"
#define DROMAJO_PLIC_BASE "0xC000000"
#define DROMAJO_PLIC_SIZE "0x4000000"
#define DROMAJO_CLINT_BASE "0x2000000"
#define DROMAJO_CLINT_SIZE "0x10000"

#endif
```

jerryz123, 2021-07-16:
This mmio thing is used for all chipyard systems now, the register holds
the value of the boot address. It’s called BootAddrReg in testchipip I
think.

I believe dromajo did not expose enough of an interface for me to pass that
address to the cosimulator when I tried to make it work. Hacking dromajo
was necessary, if unfortunate.

kabylkas, 2021-10-18:
Yeah, I will do it.