ucb-bar/chipyard #919: Repository for submodule vlsi/hammer-cadence-plugins not exist

chucklu, 2021-07-12:
Have you hidden the repository mentioned in file .gitmodules？
https://github.com/ucb-bar/chipyard/blob/master/.gitmodules#L89

> [submodule "vlsi/hammer-cadence-plugins"]
> 	path = vlsi/hammer-cadence-plugins
> 	url = https://github.com/ucb-bar/hammer-cadence-plugins.git

harrisonliew, 2021-07-12:
You need to specially request permissions to the CAD plugin repositories, as mentioned here: https://chipyard.readthedocs.io/en/latest/VLSI/Basic-Flow.html

closed 2021-07-12 by harrisonliew

chucklu, 2021-07-15:
In order to use the Hammer VLSI flow, you need access to Hammer tools and technology plugins. You can obtain these by emailing hammer-plugins-access@lists.berkeley.edu with a request for which plugin(s) you would like access to. Make sure your email includes your github ID and proof (through affiliation or otherwise) that you have licensed access to relevant tools.

liaojie321, 2021-07-15:
请问github ID和证明要怎么写

chucklu, 2021-07-15:
@liaojie321  https://github.com/ucb-bar/hammer/blob/master/doc/Hammer-Basics/Hammer-Overview.rst

Note
Unless you are a UCB BAR or BWRC affiliate or have set up a 3-way technology NDA with us, we cannot share pre-built technology plugins.

Note
If you are not a UCB BAR or BWRC affiliate and have access to tools from a specific vendor, please email hammer-plugins-access@lists.berkeley.edu with a request for which plugin(s) you would like access to. MAKE SURE TO INCLUDE YOUR GITHUB ID IN YOUR EMAIL AND YOUR ASSOCIATION TO SHOW YOU HAVE LICENSED ACCESS TO THOSE TOOLS. There will be no support guarantee for the plugin repositories, but users are encouraged to file issues and contribute patches where needed.

liaojie321, 2021-07-15:
我已经有candence工具 比如 innovus和genus请问我应该怎么做呢