ucb-bar/chipyard #923: Push ASAP7 SRAM Macros into the P&R Flow

baichen318, 2021-07-20:
Hi all,

I use ASAP7 SRAM macros (https://github.com/ucb-bar/hammer/tree/master/src/hammer-vlsi/technology/asap7) to synthesize a TinyRocket SoC (https://chipyard.readthedocs.io/en/latest/VLSI/Tutorial.html). (I can use the synthesized TinyRocketConfig SoC to simulate with different benchmarks.)

I want to push TinyRocketConfig SoC to P&R flow with **Innovus v17.11-s080_1**.

I have placed the generated SRAM macros into my floorplan using `vlsi.inputs.placement_constraints`, and the generated floorplan.tcl is attached as shown in Fig.
[image]

Nevertheless, Innovus outputs an error when it executes the command: `edit_pin -fixed_pin -pin * -hinst ChipTop -pattern fill_optimised -layer { M5 M7 } -side bottom -start { 3000 0 } -end { 0 0 }` and the error is shown below .
[image]

The error says `Layer M10: Pitch=11520x9 is still less than min width=32000 + min spacing=640.`
In the command `ccopt_design -hold -report_dir hammer_cts_debug -report_prefix hammer_cts`, Innovus outputs an induced error, `The route/traversal graph for net 'CTS_47' is not fully connected.` and terminates.
[image]

I was wondering how I can solve this problem.

Thank you all!

Best regards,
Chen

harrisonliew, 2021-07-20:
Hi Chen,

The CTS error you have shown is unrelated to the M10 issue. A clock graph that cannot be connected is usually due to excessive congestion, causing the router to be unable to connect all nodes in the clock tree. Check your floorplan and cell utilization to alleviate any hotspots.

The M10 error that you see is caused by the tech LEF from the ASAP7 PDK (this is actually the "Pad" layer, which is unused in all routing, hence it does not affect CTS).

baichen318, 2021-07-21:
> Hi Chen,
>
> The CTS error you have shown is unrelated to the M10 issue. A clock graph that cannot be connected is usually due to excessive congestion, causing the router to be unable to connect all [...]

Hi Harrison,

Thank you for your kind reply.

Do I need to rearrange the floorplan of SRAM macros in floorplan.tcl to make them as far apart as possible?

In fact, the location of every SRAM macro is decided by myself so I am afraid that I cannot find suitable locations to make Innovus run successfully. I was wondering how you solve this question?

Thank you very much, Harrison.

Best regards,
Chen

harrisonliew, 2021-07-21:
No, you just need to make sure that the SRAMs do not overlap each other or abut with their pins facing each other. The LEFs of the SRAMs contain the size of the macro, which you can use to calculate where to place them within your overall floorplan.

baichen318, 2021-07-21:
> No, you just need to make sure that the SRAMs do not overlap each other or abut with their pins facing each other. The LEFs of the SRAMs contain the size of the macro, which you can use to calculate where to place them within your overall floorplan.

Thank you for your advice, Harrison.

I have one additional question. How to make sure their pins face each other. Can I use `orientation: "r0"` for every placed SRAM macro?

Thank you very much!

Best regards,
Chen

baichen318, 2021-07-21:
I re-modify the location (i.e., x and y) in `vlsi.inputs.placement_constraints`, and the new floorplan.tcl is as shown below,
[image]

I am sure there is no overlap this time since I make gaps between different SRAM macros large enough. However, a similar error occurs but in `uartClockDomainWrapper`.
[image]

I was wondering how to solve it.

Thank you very much in advance!

Best regards,
Chen

harrisonliew, 2021-07-21:
Can you email me your innovus.log file to harrisonliew@berkeley.edu?

baichen318, 2021-07-21:
> I re-modify the location (i.e., x and y) in `vlsi.inputs.placement_constraints`, and the new floorplan.tcl is as shown below,
> [image]
>
> I am sure there is no overlap this time since I make gaps [...]

Attached is the snippet of `vlsi.inputs.placement_constraints`.
[image]

baichen318, 2021-07-23:
> Can you email me your innovus.log file to [harrisonliew@berkeley.edu](mailto:harrisonliew@berkeley.edu)?

Thanks for the help, Harrison.

The problem is resolved.

Best regards,
Chen

closed 2021-07-23 by baichen318

harrisonliew, 2021-07-23:
The resolution was to bump the Chipyard version with new don't use cells in later Hammer commits.

nishanthsmurthy24, 2023-11-09:
I ran the Default RocketChip through the ASAP7 flow and ended up with this error.

```
Start create_tracks
**WARN: (IMPFP-325):	Floorplan of the design is resized. All current create_floorplan objects are automatically derived based on specified new create_floorplan. This may change blocks, fixed standard cells, existing routes and blockages.
@file 9: # Floorplan automatically generated from HAMMER
@@file 10: place_inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_0 550 25 r0
@@file 11: create_place_halo -insts system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_0 -halo_deltas {0.0 0.0 0.0 0.0} -snap_to_site
@@file 12: create_route_halo -bottom_layer M1 -space 0.0 -top_layer M4 -inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_0
@@file 13: place_inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_1 550 270 r0
@@file 14: create_place_halo -insts system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_1 -halo_deltas {0.0 0.0 0.0 0.0} -snap_to_site
@@file 15: create_route_halo -bottom_layer M1 -space 0.0 -top_layer M4 -inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_1
@@file 16: place_inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_2 675 25 r0
@@file 17: create_place_halo -insts system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_2 -halo_deltas {0.0 0.0 0.0 0.0} -snap_to_site
@@file 18: create_route_halo -bottom_layer M1 -space 0.0 -top_layer M4 -inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_2
@@file 19: place_inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_3 675 270 r0
@@file 20: create_place_halo -insts system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_3 -halo_deltas {0.0 0.0 0.0 0.0} -snap_to_site
@@file 21: create_route_halo -bottom_layer M1 -space 0.0 -top_layer M4 -inst system/tile_prci_domain/tile_reset_domain_tile/dcache/data/data_arrays_0/data_arrays_0_ext/mem_0_3
@@file 22: place_inst system/tile_prci_domain/tile_reset_domain_tile/frontend/icache/tag_array_0/tag_array_0_ext/mem_0_0 125 150 my
**ERROR: (IMPTCM-162):	"system/tile_prci_domain/tile_reset_domain_tile/frontend/icache/tag_array_0/tag_array_0_ext/mem_0_0" does not match any object in design for specified type "inst " object in command "place_inst".

Usage: place_inst [-help] <inst> <location> [{r0 r90 r180 r270 mx mx90 my my90}] [{ -fixed  | -placed  | -soft_fixed  }]

**ERROR: (IMPTCM-3):	"system/tile_prci_domain/tile_reset_domain_tile/frontend/icache/tag_array_0/tag_array_0_ext/mem_0_0" is not a valid object value for the 1th argument.
**ERROR: (IMPSE-110):	File '/ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/floorplan.tcl' line 22: errors out.
#@ End verbose source /ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/floorplan.tcl
**ERROR: (IMPSE-110):	File '/ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/par.tcl' line 380: errors out.
#@ End verbose source /ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/par.tcl
**ERROR: (IMPSYT-6692):	Invalid return code while executing '/ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/par.tcl' was returned and script processing was stopped. Review the following error in '/ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/par.tcl' then restart.
Error info: /ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/par.tcl: 
    while executing
"place_inst system/tile_prci_domain/tile_reset_domain_tile/frontend/icache/tag_array_0/tag_array_0_ext/mem_0_0 125 150 my"
    (file "/ece/home/somas026/Project/chipyard/vlsi/build/chipyard.harness.TestHarness.RocketConfig-ChipTop/par-rundir/floorplan.tcl" line 22)
```

I am sure this is not an issue with spacing or the Macro @harrisonliew. Can you help me understand what might be the source of this problem?

Attached is the complete log file. [innovus.log](https://github.com/ucb-bar/chipyard/files/13310350/innovus.log)

harrisonliew, 2023-11-09:
The default RocketChipConfig uses a different set of SRAMs compared to the TinyRocketConfig in the tutorial. You need to find out the paths to all of the memory instances in the design and provide proper floorplan constraints for them. In this case, the instance at the path `system/tile_prci_domain/tile_reset_domain_tile/frontend/icache/tag_array_0/tag_array_0_ext/mem_0_0` does not exist in your netlist.

nishanthsmurthy24, 2023-11-09:
Thank you for that. Any tricks up the sleeve to find the paths of all memory instances? Thanking you in advance.

harrisonliew, 2023-11-09:
Unfortunately, it's not very easy. Two ways:

1. In `generated-src/<CONFIG>/` folder, inspect `top_module_hierarchy.json` combined with `gen-collateral/<CONFIG>.top.mems.v`. You can determine the path to each memory instance in the `.top.mems.v` file using the hierarchy JSON file.
2. You do synthesis, but then place and route only thru the first Innovus step (`init_design` - see the docs starting [here](https://chipyard.readthedocs.io/en/stable/VLSI/Hammer.html#vlsi-flow-control) for how to stop after a step). Open up the Innovus database using the `./generated-scripts/open_chip` in the `par-rundir`, and go to floorplan view, click on each memory instance, and note the path. IMO this is more complicated than the above, but the method above may be going away soon and you can use the GUI to drag/drop the macros around to plan the floorplan.

We haven't used Innovus' mixed placer (automatic macro placement) much yet, but it would also be an option instead of manually placing macros if the placement quality is good.