ucb-bar/chipyard #926: Can you synthesize the example?

ChristianB12, 2021-07-22:
Hi,

I'm trying to generate an SoC with Chipyard and then pass the design to OpenLane for automated RTL-to-GDS.

I'm running the simulation for the example RocketConfig and then attempting to use the generated RTL as a design for Yosys to synthesize. The issue is that (or at least I think it is) the DRAM, JTAG, Serial, and UART verilog files are all SystemVerilog that use DPI. Is there a way around this?

I've read about using hammer for VLSI flow, but I only need the design to be synthesizable so that I can run it through OpenLane.

Is what I'm trying to do feasible? Or am I completely misunderstanding some things?

lsteveol, 2021-07-24:
Which files are you trying to add for yosys synth that has the DPI? With the exception of black box resources, anything generated using Chisel should be synthesizable by default.

Are you including the DRAM verilog model that is used for simulation to yosys?

ChristianB12, 2021-07-26:
I was initially using the simulation files such as DRAM, but then realized they weren't meant for synthesis.

Now I'm running into the issue where it complains about the ClockDividerN since it isn't synthesizable Verilog

```
7.3. Executing AST frontend in derive mode using pre-parsed AST for module `\plusarg_reader'.
ERROR: Module `\ClockDividerN' referenced in module `\DividerOnlyClockGenerator' in cell `\bundleOut_0_member_allClocks_implicit_clock_clock_ClockDivideBy1' is not part of the design.
```

ChristianB12, 2021-07-26:
Okay so it seems I just had to set my top level to DigitalTop?

Could anyone explain why? I'm pretty new to this and must not understand how this is all fitting together.

lsteveol, 2021-07-28:
(Going strictly from memory) The `ChipTop` will house some fake IOs, and clock dividers. When you are just doing the `DigitalTop` you are basically generating the design minus the IOs/clk divs. 

I'm not sure why Yosys failed using the `ChipTop` as the main synth target. It's possible this is due to some of the "analog" modeling of the IOs/EICG. It's also possible that Yosys requires some of the files in order? This I'm less sure of.