ucb-bar/chipyard #929: Failed to Build the Simulator with VCS

baichen318, 2021-07-25:
Hi all,

I am failed in building the simulator with VCS with the latest SRAM behavior model (i.e., https://chipyard.readthedocs.io/en/latest/Simulation/Software-RTL-Simulation.html#choice-of-simulator).

In **sims/vcs**​ folder, I use the command, 

> make​ CONFIG=TinyRocketConfig MACROCOMPILER_MODE='-l /path/to/chipyard/vlsi/hammer/src/hammer-vlsi/technology/asap7/sram-cache.json --use-compiler --mode strict'

​​ but it is failed to build the simulator **simv-chipyard-TinyRocketConfig**​.  It outputs the error as shown below.
[image]

However, in the previous version of Chipyard, e.g., commit id: 45684f1, there is no problem.

I guess it **might be a bug** within the latest SRAM behavior model (i.e, https://github.com/ucb-bar/hammer/blob/master/src/hammer-vlsi/technology/asap7/sram_compiler/memories/behavioral/sram_behav_models.v).

My VCS version is M-2017.03 and Chipyard commit ID is b5d0131.

I was wondering how I can solve it.

Thank you all!

Best regards,
Chen

harrisonliew, 2021-07-26:
Chen,

Thanks for pointing this out! Please review ucb-bar/hammer#612 and see if it fixes it.

baichen318, 2021-07-26:
> Chen,
> 
> Thanks for pointing this out! Please review [ucb-bar/hammer#612](https://github.com/ucb-bar/hammer/pull/612) and see if it fixes it.

Hi Harrison,

Thanks for your kind reply.

I have reviewed ucb-bar/hammer#612 and I can simulate with the patch successfully now.

Thank you very much!

Best regards,
Chen

closed 2021-07-26 by baichen318