ucb-bar/chipyard #932: Intellij Build Problem ([error] Invalid build URI (no handler available):xxxx/chisel3/)

zslwyuan, 2021-08-02:
Hi all,

I am new to the mechanism of sbt and when I tried to build Chipyard in Intellij, I got the error shown below. I can use the repo in the terminal with commands but I think the features of Intellij could reduce the difficulty to navigate this HUGE project... 

Any solution will be sincerely APPRECIATED! ^_^

/usr/lib/jvm/java-1.11.0-openjdk-amd64/bin/java -Djline.terminal=jline.UnsupportedTerminal -Dsbt.log.noformat=true -Dfile.encoding=UTF-8 -Dsbt.sourcemode=true -Dsbt.workspace=$PWD/tools -Didea.managed=true -Dfile.encoding=UTF-8 -Didea.installation.dir=/snap/intellij-idea-community/315 -jar /home/server/.local/share/JetBrains/IdeaIC2021.2/Scala/launcher/sbt-launch.jar
[info] welcome to sbt 1.4.9 (Ubuntu Java 11.0.11)
[info] loading global plugins from /home/server/.sbt/1.0/plugins
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/server/Softwares/chipyard/project
[warn] There may be incompatibilities among your library dependencies; run 'evicted' to see detailed eviction warnings.
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
java.lang.RuntimeException: Invalid build URI (no handler available): file:/home/server/Softwares/chipyard/$PWD/tools/chisel3/
at scala.sys.package$.error(package.scala:30)
at sbt.internal.Load$.$anonfun$builtinLoader$1(Load.scala:483)
at sbt.internal.BuildLoader.$anonfun$apply$3(BuildLoader.scala:244)
at scala.Option.getOrElse(Option.scala:189)
at sbt.internal.BuildLoader.apply(BuildLoader.scala:244)
at sbt.internal.Load$.loadURI$1(Load.scala:548)
at sbt.internal.Load$.loadAll(Load.scala:564)
at sbt.internal.Load$.loadURI(Load.scala:494)
at sbt.internal.Load$.load(Load.scala:473)
at sbt.internal.Load$.$anonfun$apply$1(Load.scala:242)
at sbt.internal.Load$.timed(Load.scala:1382)
at sbt.internal.Load$.apply(Load.scala:242)
at sbt.internal.Load$.defaultLoad(Load.scala:55)
at sbt.BuiltinCommands$.liftedTree1$1(Main.scala:934)
at sbt.BuiltinCommands$.doLoadProject(Main.scala:934)
at sbt.BuiltinCommands$.$anonfun$loadProjectImpl$2(Main.scala:887)
at sbt.Command$.$anonfun$applyEffect$4(Command.scala:150)
at sbt.Command$.$anonfun$applyEffect$2(Command.scala:145)
at sbt.Command$.process(Command.scala:189)
at sbt.MainLoop$.$anonfun$processCommand$5(MainLoop.scala:243)
at scala.Option.getOrElse(Option.scala:189)
at sbt.MainLoop$.process$1(MainLoop.scala:243)
at sbt.MainLoop$.processCommand(MainLoop.scala:276)
at sbt.MainLoop$.$anonfun$next$5(MainLoop.scala:163)
at sbt.State$StateOpsImpl$.runCmd$1(State.scala:289)
at sbt.State$StateOpsImpl$.process$extension(State.scala:325)
at sbt.MainLoop$.$anonfun$next$4(MainLoop.scala:163)
at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
at sbt.MainLoop$.next(MainLoop.scala:163)
at sbt.MainLoop$.run(MainLoop.scala:144)
at sbt.MainLoop$.$anonfun$runWithNewLog$1(MainLoop.scala:119)
at sbt.io.Using.apply(Using.scala:27)
at sbt.MainLoop$.runWithNewLog(MainLoop.scala:112)
at sbt.MainLoop$.runAndClearLast(MainLoop.scala:66)
at sbt.MainLoop$.runLoggedLoop(MainLoop.scala:51)
at sbt.MainLoop$.runLogged(MainLoop.scala:42)
at sbt.StandardMain$.runManaged(Main.scala:193)
at sbt.xMain$.$anonfun$run$8(Main.scala:102)
at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
at scala.Console$.withIn(Console.scala:230)
at sbt.internal.util.Terminal$.withIn(Terminal.scala:560)
at sbt.internal.util.Terminal$.$anonfun$withStreams$1(Terminal.scala:350)
at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
at scala.Console$.withOut(Console.scala:167)
at sbt.internal.util.Terminal$.$anonfun$withOut$2(Terminal.scala:550)
at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
at scala.Console$.withErr(Console.scala:196)
at sbt.internal.util.Terminal$.withOut(Terminal.scala:550)
at sbt.internal.util.Terminal$.withStreams(Terminal.scala:350)
at sbt.xMain$.run(Main.scala:86)
at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
at java.base/java.lang.reflect.Method.invoke(Method.java:566)
at sbt.internal.XMainConfiguration.run(XMainConfiguration.scala:83)
at sbt.xMain.run(Main.scala:46)
at xsbt.boot.Launch$.$anonfun$run$1(Launch.scala:149)
at xsbt.boot.Launch$.withContextLoader(Launch.scala:176)
at xsbt.boot.Launch$.run(Launch.scala:149)
at xsbt.boot.Launch$.$anonfun$apply$1(Launch.scala:44)
at xsbt.boot.Launch$.launch(Launch.scala:159)
at xsbt.boot.Launch$.apply(Launch.scala:44)
at xsbt.boot.Launch$.apply(Launch.scala:21)
at xsbt.boot.Boot$.runImpl(Boot.scala:78)
at xsbt.boot.Boot$.run(Boot.scala:73)
at xsbt.boot.Boot$.main(Boot.scala:21)
at xsbt.boot.Boot.main(Boot.scala)
[error] Invalid build URI (no handler available): file:/home/server/Softwares/chipyard/$PWD/tools/chisel3/
[error] Use 'last' for the full log.

zslwyuan, 2021-08-02:
I solved the problem by changing the workspace variable in the .sbtopt file to the absolute path:
https://github.com/ucb-bar/chipyard/blob/master/.sbtopts#L2

tymcauley, 2021-08-02:
FWIW, the exact same thing happens to me when using [Metals](https://scalameta.org/metals/) (by way of [nvim-metals](https://github.com/scalameta/nvim-metals)), so I always make this same change locally (I replace the `$PWD` in `.sbtopts` with the absolute path to my Chipyard repo). The `sbt` build works fine with the `$PWD` variable in `.sbtopts`, but Metals doesn't like it.

abejgonzalez, 2021-08-23:
Noted. I'll look at the PR... but I'm not opposed to having the `.sbtopts` file generated with the absolute path to avoid this issue.

With this being said, is there any other specific setup to get Intellij working w/ Chipyard? I think a lot of new users would love to use Intellij so it might be worth adding something into our documentation?

zslwyuan, 2021-08-23:
HI @abejgonzalez ,

Thanks for your concern! ^_^ 

I can summarize some potential steps for beginners to set up Chipyard project in Intellej within these two days, and will post them in next comment for your reference.

abejgonzalez, 2021-08-23:
@tymcauley Similar to the other comment, might be worth adding some docs on how to set up Metal?

abejgonzalez, 2021-08-30:
Closing since the main issue has been resolved in #933

closed 2021-08-30 by abejgonzalez