ucb-bar/chipyard #942: Why is the git init so large?

metzkorn, 2021-08-13:
Hi, I'm starting some experimentation with chipyard and was wondering why after over half the space of the repo was in the git init? Is it being used to store any files necessary for chipyard's functionality or can I go ahead and delete it? I'm not confident on how submodules work so I'm not entirely sure if deleting it would ruin the functionality. I'm just imagining it's possible that its unmarked changes from the shell scripts and I don't think I'd need to track my changes.

jerryz123, 2021-08-17:
What file are you referring to?

metzkorn, 2021-08-17:
Sorry by "git init" I meant everything in the .git folder. If I really need to save space, I can try deleting it or using the docker image. I'm just wondering if deleting it will have side effects outside of no longer tracking changes (which end up being too large to track anyway).

closed 2021-08-19 by metzkorn