ucb-bar/chipyard #943: Top-level trait from GCD example not being generated

sammhho, 2021-08-17:
**Impact**: rtl

**Tell us about your environment:**
*Chipyard Version:* 1.5.0, also the version in the docker image
*OS:* Ubuntu 20.04, and Ubuntu 18.04.5 from the docker image
*Other:* Do `source env.sh` for custom env; No prior step for the docker image env.

**What is the current behavior?**
Under `sims/verilator`, run either 
`make CONFIG=GCDAXI4BlackBoxRocketConfig verilog` or 
`make CONFIG=GCDTLRocketConfig verilog`, 
the generated rtl in `generated-src/chipyard.TestHarness.GCDAXI4BlackBoxRocketConfig/chipyard.TestHarness.GCDAXI4BlackBoxRocketConfig.top.v`
does not have port `busy` in both `ChipTop` and `DigitalTop` level.

**What is the expected behavior?**
Seems there should be `ChipTop`/`DigitalTop` port `busy`, connected from `gcd_busy` port of `GCDTL`/`GCDAXI4`,
which is in turn connected from `busy` port of `GCDMMIOBlackBox`/`GCDMMIOChiselModule`.

**Other information**

According to docs [6.6 MMIO Peripherals/6.6.3 Top-level Traits](https://chipyard.readthedocs.io/en/latest/Customization/MMIO-Peripherals.html#top-level-traits), and also these code:
```scala
trait CanHavePeripheryGCDModuleImp extends LazyModuleImp {
  val outer: CanHavePeripheryGCD
  val gcd_busy = outer.gcd match {
    case Some(gcd) => {
      val busy = IO(Output(Bool()))
      busy := gcd.module.io.gcd_busy
      Some(busy)
    }
    case None => None
  }
}
```
seems there should be `busy` port at least on `DigitalTop` if not `ChipTop` ???
Really not sure if there's a problem in some cake pattern traits or in some part of the generator?

jerryz123, 2021-08-17:
Is the busy signal driving anything? If not, then perhaps it is getting optimized away. Try adding `dontTouch(busy)` after declaring the IO

Rodiot, 2021-08-19:
indeed  @jerryz123  is right, the signal seems to be optimized away at some point.
The `dontTouch(busy)` works, you than have the signal in the DigitalTop.
If you need it also in the ChipTop you need to add some IOBinders like so

`class WithGCDIOCells extends OverrideIOBinder({
  (system: CanHavePeripheryGCDModuleImp) => {
    val (ports, cells2d) = IOCell.generateIOFromSignal(system.gcd_busy.get, s"busy", system.p(IOCellKey), abstractResetAsAsync = true)
    (Seq(ports), cells2d)
  }
})`

Not sure if that's the intended way but it seems to work.

sammhho, 2021-08-22:
Thx @jerryz123 @Rodiot , I did try `dontTouch` and it does keep it at `DigitalTop`;
Thx for the tips on keeping it at top level too, Rodiot

jerryz123, 2021-08-23:
@Rodiot yes that is the intended mechanism for keeping it in ChipTop

closed 2021-08-23 by jerryz123

DucVu20, 2021-09-26:
Hello @sammhho, @jerryz123  and @Rodiot, I tried the code @Rodiot wrote to get the gcd_busy in the ChipTop but I got the following errors and I don't know how to fix. Do you think the code WithGCDIOCells you wrote is correct? I'm having trouble making gcd_busy available at ChipTop. For DigitalTop, I used dontTouch, and it works.
[image]

leduchuybk, 2022-12-17:
@DucVu20 I met similar problem. Have you solved it yet?

leduchuybk, 2023-06-27:
@hoangfint what version of chipyard and chisel you are using

hoangfint, 2023-06-27:
Hi @leduchuybk  
I am using chipyard 1.9.0 and chisel3