ucb-bar/chipyard #944: How to send interrupt signal to DUT during simulation in chipyard 1.5

zhuxuanlong, 2021-08-18:
Hi,
Chipyard 1.3 uses TSI interface for simulation, and the fesvr does send an interrupt signal to the DUT through the following interface:
`void htif_t::start()
{
  if (!targs.empty() && targs[0] != "none")
  {
      load_program();
  }

  reset();
}
// Interrupt core 0 to make it start executing the program in DRAM
void tsi_t::reset()
{
  uint32_t one = 1;
  write_chunk(MSIP_BASE, sizeof(uint32_t), &one);
}
`

The TSI interface is no longer used in chipyard 1.5, so how to send interrupt signal to the DUT?

Thank you

jerryz123, 2021-08-18:
Why do you say chipyard no longer uses tsi? Tsi is still used.

zhuxuanlong, 2021-08-18:
I added printing in riscv-tools\riscv-isa-sim\fesvr\tsi.cc and riscv-tools\riscv-isa-sim\fesvr\htif.cc. I can't see printing under chipyard 1.5, but chipyard 1.3 has the following printing information:
[UART] UART0 is here (stdin/stdout).
htif: run ....
htif: start ....
htif: will load_program ....
LOAD_ELF: addr:0x80000000, len:450.
tsi: write_chunk, 0x80000000.

zhuxuanlong, 2021-08-18:
Sorry, I didn't describe it clearly. I mean the scenario of quickly loading the program:
`set -o pipefail &&  ./simulator-chipyard-RocketConfig-debug +permissive +dramsim +dramsim_ini_dir=../../generators/testchipip/src/main/resources/dramsim2_ini +max-cycles=10000000 +loadmem=output/chipyard.TestHarness.RocketConfig/helloworld.loadmem_hex +loadmem_addr=80000000 --vcd=output/chipyard.TestHarness.RocketConfig/helloworld.vcd  +verbose +permissive-off /share/zhuxuanlong/MyCI/test/helloworld.riscv </dev/null 2> >(spike-dasm > output/chipyard.TestHarness.RocketConfig/helloworld.out) | tee output/chipyard.TestHarness.RocketConfig/helloworld.log
[UART] UART0 is here (stdin/stdout).
End load file.
Hello World!
`

closed 2021-08-18 by zhuxuanlong