ucb-bar/chipyard #948: Toolchain taking 2+ hours to compile. Workaround to using it?

michael-etzkorn, 2021-08-20:
My goal was to prototype on a vcu118 board with a generated risc v core. I tried setting `export MAKEFLAGS=-j8` and ran the tool script, but it's taking over 2 hours when it said it would take 20-30 minutes. Quad core i7 2.5 Ghz. Plugged in and the battery is draining. The worst part is that I think the tool chain will be unnecessary since we already have a means to load code onto a risc core. Is it normal for it to take this long? Why does the documentation claim "about 20 minutes" then? Does `export MAKEFLAGES=-j8` somehow make the process take longer? And is there a workaround to using the makefiles without the toolchain? Can I just comment out where they're used?

jerryz123, 2021-08-20:
It would be useful to know where the hangup appears to be. 2 hours seems
unreasonable for your system.

It is possible to build the RTL without the gnu toochain, but minimally
riscv-fesvr must be available for the simulators to link correctly.

michael-etzkorn, 2021-08-20:
I don't think I can identify a spot where it hangs, it seems to step forward the entire time. Sadly, some make files fail (I didn't get to save the output :/ ). I ran ubuntu-req.sh before hand. The environment is Linux Mint 20.2 Cinnamon 64-bit on VMWare Workstation Player 16. I'm just maybe out of luck with getting the toolchain up and running. Hopefully I'll be able to figure out how to leverage the RTL generator code and Vivado scripts which is all I really need for my project's purpose.

closed 2021-08-20 by michael-etzkorn

michael-etzkorn, 2021-08-20:
```
mv: (unable to get) 无法获取'/mnt/hgfs/SageProjects/chiselToys/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/dlfcn/stamp.oST' 的文件状态( (folder status) stat): 没有那个文件或目录 (don't have that folder or directory)
make[3]: *** [../o-iterator.mk:9：/mnt/hgfs/SageProjects/chiselToys/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/dlfcn/stamp.oS] 错误 1 (error)
make[3]: *** 正在等待未完成的任务.... (now waiting for other processes to complete)
make[3]: 离开目录“/mnt/hgfs/SageProjects/chiselToys/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-glibc/dlfcn”
make[2]: *** [Makefile:258：dlfcn/subdir_lib] 错误 2 (error)
make[2]: 离开目录 (leaving directory)“/mnt/hgfs/SageProjects/chiselToys/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/riscv-glibc”
make[1]: *** [Makefile:9：all] 错误 2 (error)
make[1]: 离开目录 (leaving directory)“/mnt/hgfs/SageProjects/chiselToys/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d”
make: *** [Makefile:243：stamps/build-glibc-linux-rv64imafdc-lp64d] 错误 2 (error)
```

michael-etzkorn, 2021-08-22:
Using the ariane-sdk repo to download the toolchain ended up succeeding.